zephyrproject-rtos/zephyr #19178: Segmentation fault when running echo server

rmstoi, 2019-09-16:
Our testing system noticed on Saturday there is an issue with echo server app. It segfaults when run. For example

`make pristine && cmake  -DBOARD=native_posix -DCONF_FILE=prj.conf  .. && make -j 10 && make run`

Causes:

```
UART_0 connected to pseudotty: /dev/pts/8
WARNING: Using a test - not safe - entropy source
***** Booting Zephyr OS build zephyr-v2.0.0-404-g7d27cf9f2798 *****
make[3]: *** [zephyr/CMakeFiles/run.dir/build.make:57: zephyr/CMakeFiles/run] Segmentation fault
make[2]: *** [CMakeFiles/Makefile2:501: zephyr/CMakeFiles/run.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:508: zephyr/CMakeFiles/run.dir/rule] Error 2
make: *** [Makefile:235: run] Error 2

```
Valgrind reports:

```
==29535== Thread 5:
==29535== Invalid read of size 4
==29535==    at 0x11530D: z_impl_uart_poll_out (uart.h:634)
==29535==    by 0x11530D: uart_poll_out (uart.h:55)
==29535==    by 0x11530D: write (shell_uart.c:225)
==29535==    by 0x114DC7: shell_write (shell_ops.c:380)
==29535==    by 0x11431E: shell_fprintf_buffer_flush (shell_fprintf.c:56)
==29535==    by 0x11304E: shell_raw_fprintf (shell_ops.h:27)
==29535==    by 0x11418E: shell_start (shell.c:1270)
==29535==    by 0x1141DE: shell_thread (shell.c:1183)
==29535==    by 0x111544: z_thread_entry (thread_entry.c:29)
==29535==    by 0x1174E7: posix_thread_starter (posix_core.c:305)
==29535==    by 0x4984FB8: start_thread (pthread_create.c:486)
==29535==    by 0x4A99F55: clone (clone.S:108)
==29535==  Address 0x4 is not stack'd, malloc'd or (recently) free'd

```

The same issue appears with QEMU, cmake command:

`cmake -DBOARD=qemu_x86 -DOVERLAY_CONFIG=overlay-e1000.conf ..`

rmstoi, 2019-09-16:
Priority is high because we cannot use most of networking samples e.g.:
echo server
mqtt_publisher

jukkar, 2019-09-16:
This does not seem to be native_posix related as qemu is affected too.

rmstoi, 2019-09-16:
git bisect show that causing patch is 8ce0cf0126

rmstoi, 2019-09-16:
reverting 8ce0cf0 solves the segfault for now

ulfalizer, 2019-09-16:
Think I have a fix in https://github.com/zephyrproject-rtos/zephyr/pull/19190.

PR #19190 (2019-09-16): shell: kconfig: Fix wrong UART_SHELL_ON_DEV_NAME value due to typo
The $(dt_chosen_label) preprocessor function should be passed
$(DT_CHOSEN_Z_SHELL_UART), not $(UART_SHELL_ON_DEV_NAME).

DT_CHOSEN_Z_SHELL_UART is set right before the Kconfig symbol
definition, to work around the comma in 'zephyr,shell-uart'.

The Kconfig preprocessor functions are defined in
scripts/kconfig/kconfigfunctions.py.

Fixes: #19178
commit b4094ea6 (2019-09-16): shell: kconfig: Fix wrong UART_SHELL_ON_DEV_NAME value due to typo

The $(dt_chosen_label) preprocessor function should be passed
$(DT_CHOSEN_Z_SHELL_UART), not $(UART_SHELL_ON_DEV_NAME).

DT_CHOSEN_Z_SHELL_UART is set right before the Kconfig symbol
definition, to work around the comma in 'zephyr,shell-uart'.

The Kconfig preprocessor functions are defined in
scripts/kconfig/kconfigfunctions.py.

Fixes: #19178
closed 2019-09-16 by jukkar