zephyrproject-rtos/zephyr #19181: sock_set_flag implementation in sock_internal.h does not work for 64 bit pointers

eyuval, 2019-09-16:
The implementation of sock_set_flag in subsys/net/lib/sockets/sockets_internal.h uses u32_t types and therefore does not work for 64 bit system. I am working in the native posix system:

CONFIG_BOARD="native_posix_64"
CONFIG_BOARD_NATIVE_POSIX_64BIT=y

The call to sock_set_flag happens, because I want to set the socket to O_NONBLOCK. When the sock_set_flag occurs, the ctx->user_data pointer, which previously contained a 64 bit address gets corrupted, i.e. basically downcasted to u32.

jukkar, 2019-09-16:
@eyuval can you test the fix if it works for you?

eyuval, 2019-09-16:
I tested and it fixed this specific issue. Thanks. However, I see a different problem with using the user_data pointer to store the flags. At some places e.g. in zsock_accepted_cb, the user data pointer is cast to a net_context pointer, without erasing the flag bits in the casting. Now, if one of the flags SOCK_EOF/SOCK_NONBLOCK is on, the resulting pointer will be wrong. So in my view when using the user_data pointer, one needs to discard the flag bits, or take another approach to the flag bits. Should I open a separate issue for this problem?

jukkar, 2019-09-16:
> Should I open a separate issue for this problem?

Yes, please do that.

PR #19184 (2019-09-16): net: socket: The socket flags need to use uintptr_t
The socket flags are stored in void* so we need to use uintptr_t
instead of u32_t when manipulating the flag variable.

Fixes #19181
commit c755be0e (2019-09-16): net: socket: The socket flags need to use uintptr_t

The socket flags are stored in void* so we need to use uintptr_t
instead of u32_t when manipulating the flag variable.

Fixes #19181
commit d9434250 (2019-09-25): net: socket: The socket flags need to use uintptr_t

The socket flags are stored in void* so we need to use uintptr_t
instead of u32_t when manipulating the flag variable.

Fixes #19181
closed 2019-09-16 by MaureenHelm