zephyrproject-rtos/zephyr #19186: BLE: Mesh: IVI Initiator When ivi in progress timeout

menglingao1997, 2019-09-16:
https://github.com/zephyrproject-rtos/zephyr/blob/532241d5adbfc04e61652b4ebde12c86a36cff79/subsys/bluetooth/mesh/net.c#L1351-L1375
```C
bt_mesh_beacon_ivu_initiator(true); 
```

**should check this node in primary network when initiator iv update.**

menglingao1997, 2019-09-17:
I have a question, if none of the entire Mesh nodes enable beacon `BLE_MESH_BEACON_ENABLE`, that is, there is usually no node to send the beacon. In fact, when a node performs an IVI update update, only nearby nodes(single hop) can hear this information. Unfortunately, nodes farther away will not receive this very critical information because the information cannot be effectively forwarded. 

https://github.com/zephyrproject-rtos/zephyr/blob/ab3165791bac73c773d9acc724693f547d6fe68f/subsys/bluetooth/mesh/beacon.c#L316-L335

should change just like this:
```C
 iv_change = bt_mesh_net_iv_update(iv_index, BT_MESH_IV_UPDATE(flags)); 
  
 kr_change = bt_mesh_kr_update(sub, BT_MESH_KEY_REFRESH(flags), new_key); 
 if (kr_change) { 
 	bt_mesh_net_beacon_update(sub); 
 } 
  
 if (iv_change) { 
 	/* Update all subnets */ 
 	bt_mesh_net_sec_update(NULL); 
 } else if (kr_change) { 
 	/* Key Refresh without IV Update only impacts one subnet */ 
 	bt_mesh_net_sec_update(sub); 
 } 

if (atomic_test_bit(bt_mesh.flags, BT_MESH_IVU_INITIATOR) && 
     (atomic_test_bit(bt_mesh.flags, BT_MESH_IVU_IN_PROGRESS) == 
      BT_MESH_IV_UPDATE(flags))) { 
 	bt_mesh_beacon_ivu_initiator(false); 
 } else if (iv_change || kr_change) { 
 	/*forward changed beacon information to other nodes. */ 
 	bt_mesh_beacon_ivu_initiator(true); 
 } 
```

menglingao1997, 2019-09-17:
again, I think better way to check whether node in primary netwok should implemented inside the function `bt_mesh_beacon_ivu_initiator();`
```C
bool bt_mesh_beacon_ivu_initiator(bool enable)
{
    if(!bt_mesh_subnet_get(BLE_MESH_KEY_PRIMARY))
        return false;

    bt_mesh_atomic_set_bit_to(bt_mesh.flags, BLE_MESH_IVU_INITIATOR, enable);

    if (enable) {
        k_work_submit(&beacon_timer.work);
    } else if (bt_mesh_beacon_get() == BLE_MESH_BEACON_DISABLED) {
        k_delayed_work_cancel(&beacon_timer);
    }
    return true;
}
```

jhedberg, 2019-09-17:
@menglingao1997 I think it would be better if you sent these change suggestions as a pull request. You've clearly made a lot of good analysis and having someone else implement your ideas on your behalf is not necessarily the most efficient way of going forward.

jhedberg, 2019-09-17:
> I have a question, if none of the entire Mesh nodes enable beacon BLE_MESH_BEACON_ENABLE, that is, there is usually no node to send the beacon. In fact, when a node performs an IVI update, only nearby nodes(single hop) can hear this information. Unfortunately, nodes farther away will not receive this very critical information because the information cannot be effectively forwarded.

I would say that's a badly configured network, and it's the responsibility of the network administrator to fix it, rather than something we should try to work around in the implementation. That said, I need to look into your proposed fix better to understand what exactly you're trying to do.

closed 2019-09-28 by menglingao1997