zephyrproject-rtos/zephyr #19198: Bluetooth: LL split assert on connect

aolowin, 2019-09-16:
**Describe the bug**
For a custom nRF52 board I recently switched from 1.14.99 to 2.0.99.

After a connection starts I see:
`assert: '!conn->llcp_rx' failed
`
and then:
`[00:02:37.065,216] <err> os: r0/a1:  0x00000003  r1/a2:  0x0000000a  r2/a3:  0x00000001
[00:02:37.065,216] <err> os: r3/a4:  0x0001e96a r12/ip:  0x000000f7 r14/lr:  0x00011073
[00:02:37.065,216] <err> os:  xpsr:  0x6100001b
[00:02:37.065,216] <err> os: Faulting instruction address (r15/pc): 0x0001107e
[00:02:37.065,216] <err> os: >>> ZEPHYR FATAL ERROR 3: Kernel oops
[00:02:37.065,246] <err> os: Current thread: 0x20002228 (unknown)
[00:02:37.143,127] <err> os: Halting system
`

**To Reproduce**
Compile using (the default)
CONFIG_BT_LL_SW_SPLIT

if I use the 
CONFIG_BT_LL_SW_LEGACY=y
option I everything works as usual

I'm happy to add any other tests/debug info that would be useful.

 **Other Bluetooth options in use**
CONFIG_BT=y
CONFIG_BT_DEBUG_LOG=y
CONFIG_BT_SMP=y
CONFIG_BT_PERIPHERAL=y
CONFIG_BT_CTLR_DATA_LENGTH=y
CONFIG_BT_CTLR_DATA_LENGTH_MAX=247
CONFIG_BT_CTLR_TX_BUFFER_SIZE=266
CONFIG_BT_RX_BUF_LEN=266
CONFIG_BT_L2CAP_TX_MTU=266
CONFIG_BT_GATT_DYNAMIC_DB=y
CONFIG_BT_GAP_PERIPHERAL_PREF_PARAMS=y
CONFIG_BT_PERIPHERAL_PREF_MIN_INT=6
CONFIG_BT_PERIPHERAL_PREF_MAX_INT=13
CONFIG_BT_PERIPHERAL_PREF_SLAVE_LATENCY=0
CONFIG_BT_PERIPHERAL_PREF_TIMEOUT=42

**Expected behavior**
No assertion

**Impact**
annoyance. We'll use the Legacy LL for now.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.10.3
 - Commit SHA: f646f8d0cfd08f4eebff4a3b5bc623701228e650

carlescufi, 2019-09-16:
@aolowin thanks for the report. Can you try again now that we've merged https://github.com/zephyrproject-rtos/zephyr/pull/19137 ?

aolowin, 2019-09-16:
Same result using commit: 1911a7599c9f02ced307f12f41d376307cf6db47

cvinayak, 2019-09-17:
@aolowin I have seen this recently, and could be a bug in the new controller, there has been some work done in 1.14 controller that needs porting to 2.0. I will post updates here on the fixes.

If you can use nRF sniffer, have your application use SMP debug keys, and capture the on-air traffic, it will be easy for me to reproduce and validate a fix. You may attach the logs to this issue
.

Until then if you are using the legacy controller on the upstream master, please also use:
CONFIG_BT_CTLR_ADVANCED_FEATURES=y
CONFIG_BT_TICKER_COMPATIBILITY_MODE=y

carlescufi, 2019-10-08:
@aolowin any chance you can provide sniffer logs?

aolowin, 2019-10-09:
Is this what you're looking for? Sorry for the long delay, I got pulled onto another project.
[LL-split-19198.zip](https://github.com/zephyrproject-rtos/zephyr/files/3708049/LL-split-19198.zip)

cvinayak, 2019-10-10:
@aolowin I cant find a BLE connection in your sniffer log. But, anyway, I have posted a PR, please revert back after testing with #19744

aolowin, 2019-10-10:
I've tested with https://github.com/zephyrproject-rtos/zephyr/pull/19744 and everything seems to be working correctly now. Thanks!

PR #19744 (2019-10-14): Bluetooth: controller: split: Fix ctrl proc rx node starvation
Rx nodes are reserved during control procedures for
generation of HCI event on completion. Fix the allocation
and reservation in the form of a linked list per connection
context. Worst case, this list will hold one rx node for
overlapping non-instant control procedure (Data Length
Update) plus two rx node for control procedure with instant
(PHY update with Data Length Update support).

Fixes #19198.
commit af696bdd (2019-10-14): Bluetooth: controller: split: Fix ctrl proc rx node starvation

Rx nodes are reserved during control procedures for
generation of HCI event on completion. Fix the allocation
and reservation in the form of a linked list per connection
context. Worst case, this list will hold one rx node for
overlapping non-instant control procedure (Data Length
Update) plus two rx node for control procedure with instant
(PHY update with Data Length Update support).

Fixes #19198.
closed 2019-10-14 by carlescufi