zephyrproject-rtos/zephyr #19216: Ext library for WIN1500: different values of AF_INET

apapkovskiy, 2019-09-17:
After merging dac9268a2169f35f486bf904f52657547f84da75 (net/ip: Protocol family do not have to follow any other OS's value #12854) the external library for wifi WIN1500 and zephyr net core have different values for AF_INET:
1. zephyr/include/net/net_ip.h AF_INET is 1
2. zephyr/ext/hal/atmel/asf/common/components/wifi/winc1500/socket/include/socket.h AF_INET is 1
As a result it doesn't work properly.

I suppose any "offload" library (with its own network stack) will faced with the same issue ((((

jukkar, 2019-09-17:
According to comment in `drivers/wifi/winc1500/wifi_winc1500.c`, the driver tries to avoid including winc1500 version of socket.h
```
/* We do not need <socket/include/socket.h>
 * It seems there is a bug in ASF side: if OS is already defining sockaddr
 * and all, ASF will not need to define it. Unfortunately its socket.h does
 * but also defines some NM API functions there (??), so we need to redefine
 * those here.
 */
```
Actually this is not enough as basically we have a binary interface between zephyr and offloaded stack. The offloaded driver might have been already compiled in companion chip etc so we probably cannot change the value in HAL (is this correct assumption?).
In Zephyr side, currently the family value needs to be <=8 as we are storing it in three bits in net_context.h so changing the value is also restricted.
In theory we could probably have something like this in various places
```
#if !defined(AF_INET)
#define AF_INET 1
#endif 
```
but it might not work in practice.
Probably the only safe way to support the offloaded driver, is that the offloaded driver in Zephyr does mapping from Zephyr value -> HAL value.

jukkar, 2019-09-17:
It seems that TI simplelink HAL does this correctly, it defines SLNETSOCK_AF_INET and then maps AF_INET to SL_AF_INET in `drivers/wifi/simplelink/simplelink_sockets.c`

jukkar, 2019-09-17:
To me this looks like a bug in the Atmel HAL, it should not define AF_INET in the first place as there can be a conflict between the value in the host OS and the HAL.

apapkovskiy, 2019-09-17:
@jukkar ... Yes, I think you are right, it seems an Atmel's bug. Thanks for a good example from TI. I'm trying to do the same way.

ioannisg, 2019-09-17:
@apapkovskiy could you please close this bug, if this is not a Zephyr issue?
Also, if you think this issue is present in hal/atmel repo ,could you please  open an issue there?

galak, 2019-09-17:
Here's a link to bug reporting for Atmel:

https://asf.microchip.com/docs/latest/bug_tracker.html

dleach02, 2019-09-17:
I would argue that the Zephyr definition for AF_INET is the problem. I get that this is a define but if you are going to need to pass this value over an interface than we should attempt to align with convention. The convention was established with Berkeley Sockets and replicated across many different OS/RTOS platforms that I can see. And while we are at it we should go ahead and change AF_INET6 as well.

I propose the following changes (to align with Linux):

#define PF_INET  2
#define PF_INET6  10
#define PF_CAN 29

jukkar, 2019-09-18:
> I get that this is a define but if you are going to need to pass this value over an interface than we should attempt to align with convention.

The point is that the value should not be passed as is over an binary interface because it is not standardized. People might be using some historic value for AF_INET, but it cannot be know for sure that it is the same value everywhere. So ATMEL HAL should it it properly instead of trying to dictate how the networking stack in the host system is implemented.

What the Atmel HAL is doing wrong here, is that it is defining AF_INET itself which does not work. It should not define AF_INET at all as there will be a conflict with host system. Instead it could for example define ATMEL_AF_INET which is its idea of what the value of AF_INET is. Then in the offloading wifi driver we can map our understanding of AF_INET to Atmel's one. This is done properly in TI's SimpleLink driver.

cc: @tbursztyka

apapkovskiy, 2019-09-18:
Partially agree with @jukkar. But Atmel HAL is an external library and I support the notion it is the zephyr win1500 driver responsibility to map this values. As
```
/* We do not need <socket/include/socket.h>
 * It seems there is a bug in ASF side: if OS is already defining sockaddr
 * and all, ASF will not need to define it. Unfortunately its socket.h does
 * but also defines some NM API functions there (??), so we need to redefine
 * those here.
 */
```
The driver should introduce  ATMEL_AF_INET and then uses it to pass to Atmel HAL

jukkar, 2019-09-18:
> The driver should introduce ATMEL_AF_INET and then uses it to pass to Atmel HAL

That could be done, but it would be only a short term workaround. Ultimately the Atmel HAL needs to be fixed.

apapkovskiy, 2019-09-23:
After discussion I think it is not a bug, I'm not sure it is the Atmel HAL needs to be fixed but it is outside the topic.

closed 2019-09-23 by apapkovskiy