zephyrproject-rtos/zephyr #19226: Device Tree Enhancements in 2.1

galak, 2019-09-17:


galak, 2019-09-17:
[ ] Remove need of .conf file from DT used by Kconfig
[ ] Replace use of .conf from Sanitycheck
[ ] Updates to dt-binding format: simplify 'compatible' / constraint matching
[ ] Updates to dt-binding format: simplify 'include'
[ ] Updates to dt-binding format: simplify 'parent-bus' / 'child-bus'
[ ] Updates to dt-binding format: simplify 'include'
[ ] Updates to dt-binding format: category -> required

ulfalizer, 2019-09-17:
Working on a `child-binding:` thing now too. Works like `sub-node:`, except you put an complete binding in there, and it it works recursively too, so that you can spec stuff like this in the `foo-grandchildren` binding.

```
parent {
    compatible = "foo-grandchildren";
    child-1 {
        grandchild-1 {
            foo = <1>;
        };
        grandchild-2 {
            foo = <2>;
        };
        ...
    };
    child-2 {
        grandchild {
            foo = <1>;
        };
    };
};
```

@b0661 wanted it for some pinmux thing.

galak, 2019-11-04:
PR #20319 is the last in tree user of generated_dts_board.conf

closed 2019-11-05 by galak