zephyrproject-rtos/zephyr #19238: tests/subsys/usb/device failed on reel_board.

chen-png, 2019-09-18:
**Describe the bug**
tests/subsys/usb/device failed on reel_board, this issue occurred after the commit ID 3fe53db5c.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b=reel_board tests/subsys/usb/device
2. west flash

**Screenshots or console output**
***** Booting Zephyr OS build zephyr-v2.0.0-474-ga077bdd417d5 *****
[00:00:00.004,882] [0m<dbg> usb_descriptor.usb_get_device_descriptor: __usb_descriptor_start 0x20003400[0m
[00:00:00.014,373] [0m<dbg> usb_descriptor.usb_get_device_descriptor: __usb_descriptor_end 0x20003459[0m
ASSERTION FAIL [out_ctx->control_block->offset <= out_ctx->size] @ ZEPHYR_BASE/subsys/logging/log_output.c:101
ASSERTION FAIL [out_ctx->control_block->offset <= out_ctx->size] @ ZEPHYR_BASE/subsys/logging/log_output.c:101

**Environment (please complete the following information):**
 - OS: Fedora28
 - Toolchain: Zephyr-sdk-0.10.3
 - Commit ID: a077bdd417d5

xuhao8210, 2019-09-18:
Added some additional information :

I found that the commit id : 2192499fe0e8951c721e73bb58239da364fd6770 caused to the failure,and there are more failure cases below on sam_e70_xplained platform due to same error. After I disabled the TEST_LOGGING_DEFAULTS optional arguments (added one row:”CONFIG_TEST_LOGGING_DEFAULTS=n” in the file:”prj.conf” of the failure case), the case can be passed.

  sam_e70_xplained          tests/net/ip-addr/net.ip-addr                      FAILED: timeout (device)
  sam_e70_xplained          tests/net/socket/register/net.socket.register      FAILED: timeout (device)
  sam_e70_xplained          tests/net/iface/net.iface.no_userspace_allowed     FAILED: timeout (device)
  sam_e70_xplained          tests/net/ptp/clock/net.ptp.clock                  FAILED: timeout (device)
  sam_e70_xplained          tests/net/trickle/net.trickle                      FAILED: timeout (device)
  sam_e70_xplained          tests/net/iface/net.iface.with_userspace           FAILED: timeout (device)
  sam_e70_xplained          tests/net/tx_timestamp/net.timestamp               FAILED: timeout (device)
  sam_e70_xplained          tests/subsys/usb/device/usb.device                 FAILED: timeout (device)
  sam_e70_xplained          tests/subsys/usb/desc_sections/usb.desc_sections   FAILED: timeout (device)

xuhao8210, 2019-09-18:
@andrewboie could you please help have a look firstly ? Thanks

jfischer-phytec-iot, 2019-09-18:
LOG_IMMEDIATE must not be used with USB subsystem.

andrewboie, 2019-11-05:
Is this still a problem?

chen-png, 2019-11-05:
> Is this still a problem?

It didn't report this issue any more. but in the #19339 it said that it related to this issue but didn't fix this issue, so I didn't close this.

closed 2019-11-12 by andrewboie