zephyrproject-rtos/zephyr #19239: tests/kernel/common failed on iotdk board.

chen-png, 2019-09-18:
**Describe the bug**
tests/kernel/common failed on iotdk board, this issue occurred after the commit ID 3fe53db.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b=iotdk tests/kernel/common
2. west flash

**Screenshots or console output**
***** delaying boot 500ms (per build configuration) *****
***** Booting Zephyr OS build zephyr-v2.0.0-474-ga077bdd417d5 (delayed boot 500ms) *****
Running test suite common
starting test - test_verify_bootdelay
PASS - test_verify_bootdelay

starting test - test_irq_offload
PASS - test_irq_offload

starting test - test_byteorder_memcpy_swap
PASS - test_byteorder_memcpy_swap

starting test - test_byteorder_mem_swap
PASS - test_byteorder_mem_swap

starting test - test_sys_get_be64
PASS - test_sys_get_be64

starting test - test_sys_put_be64
PASS - test_sys_put_be64

starting test - test_sys_get_be32
PASS - test_sys_get_be32
starting test - test_sys_put_be32
PASS - test_sys_put_be32

starting test - test_sys_get_be16
PASS - test_sys_get_be16

starting test - test_sys_put_be16
PASS - test_sys_put_be16

starting test - test_sys_get_le16
PASS - test_sys_get_le16

starting test - test_sys_put_le16
PASS - test_sys_put_le16

starting test - test_sys_get_le32
PASS - test_sys_get_le32

starting test - test_sys_put_le32
PASS - test_sys_put_le32

starting test - test_sys_get_le64
PASS - test_sys_get_le64

starting test - test_sys_put_le64
PASS - test_sys_put_le64

starting test - test_atomic
Unimplemented system call
[00:00:00.717,000] [1;31m<err> os: >>> ZEPHYR FATAL ERROR 3: Kernel oops[0m
[00:00:00.724,000] [1;31m<err> os: Current thread: 0x80000df0 (unknown)[0m
[00:00:00.731,000] [1;31m<err> os: Halting system[0m

**Environment (please complete the following information):**
 - OS: Fedora28
 - Toolchain: Zephyr-sdk-0.10.3
 - Commit ID: a077bdd

andrewboie, 2019-09-18:
What implementation of atomic operations is this board using?

maksimmasalski, 2019-09-25:
@chen-png are you sure that problem started to happen after that commit #3fe53db ? Because it seems that commit wasn't connected with any of the core changes of the kernel.

maksimmasalski, 2019-09-25:
@andrewboie @ioannisg 
Do you have any concerns where is the problem? Today I tried to use GDB to debug it, and program crashes in line 47 of the file atomic.c `atomic_add` function.
I commented that function out from the file atomic.c, 
`	/* atomic_add() */
	target = 1;
	value = 2;
	zassert_true((atomic_add(&target, value) == 1), "atomic_add");
	zassert_true((target == 3), "atomic_add");`
and also deleted call of that function from the main.c file.
Then I ran a test again, it was passed successfully. 
I decided to dig deeper, opened file zephyr/kernel/atomic_c.c, checked how that function is implemented. 
atomic_add is not working (we receive that error "Unimplemented system call"), but atomic_sub works well, logic of that functions is absolutely the same and code for that functions is the same except one function has "+", another has "-" in the code.
I also tried to change setting in prj.conf file and changed CONFIG_TEST_USERSPACE setup, but still the same error.
Any thoughts about it?

maksimmasalski, 2019-09-26:
@vonhust Let's try to solve that issue together.

vonhust, 2019-09-26:
@maksimmasalski the root cause of this issue is #19370. I will provide a fix soon

PR #19401 (2019-09-27): kernel: fix the bug in atomic_c.c
* USERSPACE -> CONFIG_USERSPACE
* fix the wrong paramter type

Fixes #19370 
Fixes #19239
closed 2019-09-27 by nashif