zephyrproject-rtos/zephyr #19243: FRDM-K64F samples/subsys/fs example failed to compile

Lowentwickler, 2019-09-18:
**Describe the bug**
Comilation of samples/subsys/fs example can not be done due to missing SPI definitions

**To Reproduce**
`west build -b frdm_k64f samples/subsys/fs`

**Expected behavior**
Sample application must be compiled

**Impact**
Blocker

**Screenshots or console output**
`
[ 88%] Building C object zephyr/drivers/spi/CMakeFiles/drivers__spi.dir/spi_mcux_dspi.c.obj
In file included from /home/elias/zephyrproject/zephyr/include/generated_dts_board.h:18,
                 from /home/elias/zephyrproject/zephyr/include/arch/arm/arch.h:20,
                 from /home/elias/zephyrproject/zephyr/include/arch/cpu.h:17,
                 from /home/elias/zephyrproject/zephyr/include/kernel_includes.h:34,
                 from /home/elias/zephyrproject/zephyr/include/kernel.h:17,
                 from /home/elias/zephyrproject/zephyr/include/device.h:11,
                 from /home/elias/zephyrproject/zephyr/include/drivers/spi.h:24,
                 from /home/elias/zephyrproject/zephyr/drivers/spi/spi_mcux_dspi.c:9:
/home/elias/zephyrproject/zephyr/build/zephyr/include/generated/generated_dts_board_fixups.h:116:32: error: 'DT_NXP_KINETIS_DSPI_4002D000_BASE_ADDRESS' undeclared here (not in a function); did you mean 'DT_NXP_KINETIS_DSPI_4002C000_BASE_ADDRESS'?
  116 | #define DT_SPI_1_BASE_ADDRESS  DT_NXP_KINETIS_DSPI_4002D000_BASE_ADDRESS
      |                                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/elias/zephyrproject/zephyr/build/zephyr/include/generated/generated_dts_board_fixups.h:116:32: note: in definition of macro 'DT_SPI_1_BASE_ADDRESS'
  116 | #define DT_SPI_1_BASE_ADDRESS  DT_NXP_KINETIS_DSPI_4002D000_BASE_ADDRESS
      |                                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/elias/zephyrproject/zephyr/build/zephyr/include/generated/generated_dts_board_fixups.h:119:31: error: 'DT_NXP_KINETIS_DSPI_4002D000_CLOCK_CONTROLLER' undeclared here (not in a function); did you mean 'DT_NXP_KINETIS_DSPI_4002C000_CLOCK_CONTROLLER'?
  119 | #define DT_SPI_1_CLOCK_NAME   DT_NXP_KINETIS_DSPI_4002D000_CLOCK_CONTROLLER
      |                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/elias/zephyrproject/zephyr/build/zephyr/include/generated/generated_dts_board_fixups.h:119:31: note: in definition of macro 'DT_SPI_1_CLOCK_NAME'
  119 | #define DT_SPI_1_CLOCK_NAME   DT_NXP_KINETIS_DSPI_4002D000_CLOCK_CONTROLLER
      |                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/elias/zephyrproject/zephyr/build/zephyr/include/generated/generated_dts_board_fixups.h:120:32: error: 'DT_NXP_KINETIS_DSPI_4002D000_CLOCK_NAME' undeclared here (not in a function); did you mean 'DT_NXP_KINETIS_DSPI_4002C000_CLOCK_NAME'?
  120 | #define DT_SPI_1_CLOCK_SUBSYS  DT_NXP_KINETIS_DSPI_4002D000_CLOCK_NAME
      |                                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/elias/zephyrproject/zephyr/build/zephyr/include/generated/generated_dts_board_fixups.h:120:32: note: in definition of macro 'DT_SPI_1_CLOCK_SUBSYS'
  120 | #define DT_SPI_1_CLOCK_SUBSYS  DT_NXP_KINETIS_DSPI_4002D000_CLOCK_NAME
      |                                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/elias/zephyrproject/zephyr/include/drivers/spi.h:24,
                 from /home/elias/zephyrproject/zephyr/drivers/spi/spi_mcux_dspi.c:9:
/home/elias/zephyrproject/zephyr/build/zephyr/include/generated/generated_dts_board_fixups.h:115:25: error: 'DT_NXP_KINETIS_DSPI_4002D000_LABEL' undeclared here (not in a function); did you mean 'DT_NXP_KINETIS_DSPI_4002C000_LABEL'?
  115 | #define DT_SPI_1_NAME   DT_NXP_KINETIS_DSPI_4002D000_LABEL
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/elias/zephyrproject/zephyr/include/device.h:107:11: note: in definition of macro 'DEVICE_AND_API_INIT'
  107 |   .name = drv_name, .init = (init_fn),     \
      |           ^~~~~~~~
/home/elias/zephyrproject/zephyr/drivers/spi/spi_mcux_dspi.c:314:33: note: in expansion of macro 'DT_SPI_1_NAME'
  314 | DEVICE_AND_API_INIT(spi_mcux_1, DT_SPI_1_NAME, &spi_mcux_init,
      |                                 ^~~~~~~~~~~~~
In file included from /home/elias/zephyrproject/zephyr/include/arch/arm/cortex_m/irq.h:18,
                 from /home/elias/zephyrproject/zephyr/include/arch/arm/arch.h:27,
                 from /home/elias/zephyrproject/zephyr/include/arch/cpu.h:17,
                 from /home/elias/zephyrproject/zephyr/include/kernel_includes.h:34,
                 from /home/elias/zephyrproject/zephyr/include/kernel.h:17,
                 from /home/elias/zephyrproject/zephyr/include/device.h:11,
                 from /home/elias/zephyrproject/zephyr/include/drivers/spi.h:24,
                 from /home/elias/zephyrproject/zephyr/drivers/spi/spi_mcux_dspi.c:9:
/home/elias/zephyrproject/zephyr/drivers/spi/spi_mcux_dspi.c: In function 'spi_mcux_config_func_1':
/home/elias/zephyrproject/zephyr/build/zephyr/include/generated/generated_dts_board_fixups.h:117:24: error: 'DT_NXP_KINETIS_DSPI_4002D000_IRQ_0' undeclared (first use in this function); did you mean 'DT_NXP_KINETIS_DSPI_4002C000_IRQ_0'?
  117 | #define DT_SPI_1_IRQ   DT_NXP_KINETIS_DSPI_4002D000_IRQ_0
      |                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/elias/zephyrproject/zephyr/include/sw_isr_table.h:70:5: note: in definition of macro 'Z_ISR_DECLARE'
   70 |    {irq, flags, &func, (void *)param}
      |     ^~~
/home/elias/zephyrproject/zephyr/include/irq.h:51:2: note: in expansion of macro 'Z_ARCH_IRQ_CONNECT'
   51 |  Z_ARCH_IRQ_CONNECT(irq_p, priority_p, isr_p, isr_param_p, flags_p)
      |  ^~~~~~~~~~~~~~~~~~
[... 25 lines cut ...]
                 from /home/elias/zephyrproject/zephyr/include/device.h:11,
                 from /home/elias/zephyrproject/zephyr/include/drivers/spi.h:24,
                 from /home/elias/zephyrproject/zephyr/drivers/spi/spi_mcux_dspi.c:9:
/home/elias/zephyrproject/zephyr/build/zephyr/include/generated/generated_dts_board_fixups.h:118:28: error: 'DT_NXP_KINETIS_DSPI_4002D000_IRQ_0_PRIORITY' undeclared (first use in this function); did you mean 'DT_NXP_KINETIS_DSPI_4002C000_IRQ_0_PRIORITY'?
  118 | #define DT_SPI_1_IRQ_PRI   DT_NXP_KINETIS_DSPI_4002D000_IRQ_0_PRIORITY
      |                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/elias/zephyrproject/zephyr/include/arch/arm/cortex_m/irq.h:85:28: note: in definition of macro 'Z_ARCH_IRQ_CONNECT'
   85 |  z_irq_priority_set(irq_p, priority_p, flags_p); \
      |                            ^~~~~~~~~~
/home/elias/zephyrproject/zephyr/drivers/spi/spi_mcux_dspi.c:321:2: note: in expansion of macro 'IRQ_CONNECT'
  321 |  IRQ_CONNECT(DT_SPI_1_IRQ, DT_SPI_1_IRQ_PRI,
      |  ^~~~~~~~~~~
/home/elias/zephyrproject/zephyr/drivers/spi/spi_mcux_dspi.c:321:28: note: in expansion of macro 'DT_SPI_1_IRQ_PRI'
  321 |  IRQ_CONNECT(DT_SPI_1_IRQ, DT_SPI_1_IRQ_PRI,
      |                            ^~~~~~~~~~~~~~~~
make[2]: *** [zephyr/drivers/spi/CMakeFiles/drivers__spi.dir/build.make:63: zephyr/drivers/spi/CMakeFiles/drivers__spi.dir/spi_mcux_dspi.c.obj] Chyba 1
make[1]: *** [CMakeFiles/Makefile2:2058: zephyr/drivers/spi/CMakeFiles/drivers__spi.dir/all] Chyba 2
make: *** [Makefile:84: all] Chyba 2
ERROR: command exited with status 2: /usr/bin/cmake --build /home/elias/zephyrproject/zephyr/build
`

**Environment (please complete the following information):**
 - Archlinux 5.2.11-arch1-1-ARCH
 - Toolchain arm-none-eabi-gcc (Arch Repository) 9.2.0
 - commit 765c06376c12ad332d792fa19a6f4b40d2841ad0

galak, 2019-09-18:
If you look at `samples/subsys/fs/sample.yaml` you'll see the platforms that this sample is supported on.  The FRDM-K64F is not one of them.  I don't know if anyone has tried SDHC support on the FRDM-K64F.

Lowentwickler, 2019-09-18:
I need to make it work. Any guidance is very welcome.

explora26, 2019-09-18:
The SPI 1 is not enable in device tree by default.
You need add the frdm_k64f.overlay in samples/subsys/fs/fat_fs
```
&spi1 {
        status = "okay";
};
```

Lowentwickler, 2019-09-18:
Yes, I already tried this. Application is being compiled. But SD mount fails with:
```
[00:00:08.506,000] [1;31m<err> main: Storage init ERROR![0m
[00:00:08.506,000] [1;31m<err> fs: fs mount error (-5)[0m
Error mounting disk.
```

galak, 2019-09-18:
@MaureenHelm do you know if anyone's tried SDHC on FRDM-K64F.

Lowentwickler, 2019-09-18:
It seems that I also need to define some more in DTS:
```
&spi1 {
	status = "okay";
	cs-gpios = <&gpioe 5 0>;
	sdhc0: sdhc@0 {
		compatible = "zephyr,mmc-spi-slot";
		reg = <0>;
		status = "okay";
		label = "SDHC0";
		spi-max-frequency = <25000000>;
	};
};
```
I am not sure about value for gpio-cs. And still can not find where I should define pins for SPI1.

MaureenHelm, 2019-09-18:
> @MaureenHelm do you know if anyone's tried SDHC on FRDM-K64F.

Not that I'm aware of. @JunYangNXP does k64 have the same sdhc controller as i.mx rt?

JunYangNXP, 2019-09-19:
@MaureenHelm I see the SDHC block of K6x, but didn't find detailed reference manual. Anyone can share this Doc?

Lowentwickler, 2019-09-19:
@JunYangNXP [Reference Manual](https://github.com/Lowentwickler/frdm-k64f/blob/master/K64P144M120SF5RM.pdf) see chapter 53

JunYangNXP, 2019-09-20:
Seems the SDHC of K64 is subset of uSDHC of RT, Zephyr usdhc driver is supposed to support this by additional changes and some configurations.
I am wondering if the SD interface of FRDM-K64F is controlled by this SDHC or SPI?

Lowentwickler, 2019-09-20:
@JunYangNXP you can find [Schematic](https://github.com/Lowentwickler/frdm-k64f/blob/master/FRDM-K64F-SCH-E4.pdf) It seems it is controlled over SDHC.

JunYangNXP, 2019-09-20:
@Lowentwickler Then you need to make uSDHC driver working for FRDM-K64F, which does not depends on SPI. You can refer the MIMXRT1050 configuration for this. But  uSDHC driver is supposed to change to work for K64.

Lowentwickler, 2019-09-20:
@JunYangNXP definitions from `../modules/hal/nxp/mcux/devices/MK64F12/MK64F12.h` are different than in `../modules/hal/nxp/mcux/devices/MIMXRT1052/MIMXRT1052.h`.

For example:
```
USDHC_PRES_STATE_CIHB_MASK <>  SDHC_PRSSTAT_CIHB_MASK
```

It is not hard to find proper analogs, but it seems wrong way for me to correct it in MK64F12.h, neither in disk_access_usdhc.c. What is the Zephyr way for this situation? Make a copy of disk_access_usdhc.c and make corrections there? Or make new header file with defines and typedefs that will be included in disk_access_usdhc.c? Or whatever?

JunYangNXP, 2019-09-20:
The sdhc of K64 is subset of that of RT, so the driver is not exactly compatible to K64 and needs further development. Any sdhc relevant definition in soc header is supposed to be moved driver header.

galak, 2019-09-20:
I'd suggest maybe making whatever hacks to the driver you need to make it work.  If you want to post that as a draft PR we can probably than provide guidance on how to clean up the driver/etc to get it acceptable upstream.

Lowentwickler, 2019-09-22:
I am debugging my driver for kinetis SDHC and getting this on first attept to read SYSCTRL register (0x400b102c):
```
[00:00:00.000,000] [0m<dbg> disk.disk_access_register: disk interface(SD) registred[0m
[00:00:00.000,000] [0m<inf> usdhc: SD inserted!
[0m
[00:00:00.000,000] [1;31m<err> os: ***** HARD FAULT *****[0m
[00:00:00.000,000] [1;31m<err> os:   Fault escalation (see below)[0m
[00:00:00.000,000] [1;31m<err> os: ***** BUS FAULT *****[0m
[00:00:00.000,000] [1;31m<err> os:   Precise data bus error[0m
[00:00:00.000,000] [1;31m<err> os:   BFAR Address: 0x400b102c[0m
[00:00:00.000,000] [1;31m<err> os: r0/a1:  0x20002298  r1/a2:  0x20004018  r2/a3:  0x00000000[0m
[00:00:00.000,000] [1;31m<err> os: r3/a4:  0x00000008 r12/ip:  0x0000000f r14/lr:  0x00001103[0m
[00:00:00.000,000] [1;31m<err> os:  xpsr:  0x41000005[0m
[00:00:00.000,000] [1;31m<err> os: Faulting instruction address (r15/pc): 0x0000222a[0m
[00:00:00.000,000] [1;31m<err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception[0m
[00:00:00.000,000] [1;31m<err> os: Current thread: 0x20002720 (unknown)[0m
[00:00:00.372,000] [1;31m<err> os: Halting system[0m
```
Is it because there is something missing in Zephyr configuration?
I've added this to nxp_k6x.dtsi:
```
		usdhc1: usdhc@400b1000 {
			compatible = "nxp,imx-usdhc";
			reg = <0x400b1000 0x4000>;
			status = "disabled";
			interrupts = <76 0>;
			clocks = <&sim KINETIS_SIM_CORESYS_CLK 0x1034 12>;
			label = "USDHC_1";

			miso-mosi-clk {
				pins = <1>, <3>, <2>;
				function = <2>;
			};
		};
```
and this to prj_frdm_k64f.conf:
```
CONFIG_GPIO=y
CONFIG_GPIO_MCUX_PORTE=y
CONFIG_PINMUX_MCUX=y
CONFIG_PINMUX_MCUX_PORTE=y
```
Also to pinmux.c:
```
	pinmux_pin_set(porte, 0, PORT_PCR_MUX(kPORT_PullUp) | PORT_PCR_MUX(kPORT_MuxAlt4));
	pinmux_pin_set(porte, 1, PORT_PCR_MUX(kPORT_PullUp) | PORT_PCR_MUX(kPORT_MuxAlt4));
	pinmux_pin_set(porte, 2, PORT_PCR_MUX(kPORT_PullUp) | PORT_PCR_MUX(kPORT_MuxAlt4));
	pinmux_pin_set(porte, 3, PORT_PCR_MUX(kPORT_PullUp) | PORT_PCR_MUX(kPORT_MuxAlt4));
	pinmux_pin_set(porte, 4, PORT_PCR_MUX(kPORT_PullUp) | PORT_PCR_MUX(kPORT_MuxAlt4));
	pinmux_pin_set(porte, 5, PORT_PCR_MUX(kPORT_PullUp) | PORT_PCR_MUX(kPORT_MuxAlt4));
	pinmux_pin_set(porte, 6, PORT_PCR_MUX(kPORT_PullUp) | PORT_PCR_MUX(kPORT_MuxAsGpio));
```
I am new to Zephyr and don't understand some configuration concepts, so any recommendations are very welcome.

henrikbrixandersen, 2022-11-02:
SD/MMC support was added to the `frdm_k64f` board in 33b851f71e4268685e850f053d7577026f052b53. Can this be closed?

Lowentwickler, 2022-11-02:
I think so

closed 2022-11-02 by henrikbrixandersen