zephyrproject-rtos/zephyr #19250: NVS: Overwriting an item with a shorter matching item fails

jbrzozoski, 2019-09-18:
**Describe the bug**
Calling nvs_write to overwrite a data item with another data item that is a truncated version of the existing data will not overwrite the old item.  The "is this equal" check in nvs_write ignores differing lengths and then compares the old/new data up to the length of the incoming item, but it should consider differing lengths as non-matching.

**To Reproduce**
Steps to reproduce the behavior:
1. Call nvs_write and store a string like "ABCD"
2. Call nvs_write again to overwrite the item with "ABC"
3. Read back the NVS item and get "ABCD"

**Expected behavior**
In step 3 above, you should read back "ABC"

**Impact**
Could be causing subtle issues for any application that uses NVS.  Workarounds are possible if you're aware of the problem.

**Additional context**
A simple patch to fix this is attached.

[nvs_patch.diff.txt](https://github.com/zephyrproject-rtos/zephyr/files/3627146/nvs_patch.diff.txt)

Laczen, 2019-09-18:
@jbrzozoski, thanks for reporting this. Your analysis is correct as well as the proposed fix.

carlescufi, 2019-09-18:
@jbrzozoski thanks for the patch, could you actually open a Pull Request with that code?

PR #19254 (2019-09-19): nvs: fix overwriting with truncated data
When overwriting an NVS item with data that was a truncated version of
the existing data, the "is this already saved" logic was ignoring the
differing lengths and not saving the new item because the data matched.

Fixes #19250
commit d0a368e2 (2019-09-19): nvs: fix overwriting with truncated data

When overwriting an NVS item with data that was a truncated version of
the existing data, the "is this already saved" logic was ignoring the
differing lengths and not saving the new item because the data matched.

Fixes #19250
closed 2019-09-19 by nashif