zephyrproject-rtos/zephyr #19263: Bluetooth: Mesh: Friend Clear Procedure Timeout

menglingao1997, 2019-09-19:
https://github.com/zephyrproject-rtos/zephyr/blob/99e935e2ba740c309d43a3608352e6bce494662e/subsys/bluetooth/mesh/friend.c#L647-L673

may be overflow:
```C
// unsigned type
duration = k_uptime_get_32() - frnd->clear.start; 
 	if (duration > 2 * frnd->poll_to) {
```

due to init as:
```C
frnd->clear.start = k_uptime_get_32() + (2 * frnd->poll_to);
```

again, mesh spec define the  friend clear procedure timeout as 2*poll_timeout.

jhedberg, 2019-09-19:
It's almost two years ago I wrote this code, but to me it looks like the `frnd->clear.start` should simply be initialised to `k_uptime_get_32()` rather than what the code now does. Do you agree?

Regarding integer overflow, as long as we don't go past the same value as the start timestamp after wrapping around to 0, the subtraction will still yield the correct result.

menglingao1997, 2019-09-19:
> It's almost two years ago I wrote this code, but to me it looks like the `frnd->clear.start` should simply be initialised to `k_uptime_get_32()` rather than what the code now does. Do you agree? [...]

sure

PR #19266 (2019-09-26): Bluetooth: Mesh: Fix Clear Procedure start timestamp initialization
The start timestamp was supposed to signify the starting point of the
clear procedure. The code was incorrectly initializing it to the *end*
point of the procedure.

Fixes #19263
commit 6d798c8c (2019-09-26): Bluetooth: Mesh: Fix Clear Procedure start timestamp initialization

The start timestamp was supposed to signify the starting point of the
clear procedure. The code was incorrectly initializing it to the *end*
point of the procedure.

Fixes #19263
commit 451f004f (2019-09-26): Bluetooth: Mesh: Fix Clear Procedure start timestamp initialization

The start timestamp was supposed to signify the starting point of the
clear procedure. The code was incorrectly initializing it to the *end*
point of the procedure.

Fixes #19263
commit 30495ab7 (2019-09-26): Bluetooth: Mesh: Fix Clear Procedure start timestamp initialization

The start timestamp was supposed to signify the starting point of the
clear procedure. The code was incorrectly initializing it to the *end*
point of the procedure.

Fixes #19263
closed 2019-09-26 by jhedberg