zephyrproject-rtos/zephyr #19265: Bluetooth: Mesh: Friend Send model message to LPN

menglingao1997, 2019-09-19:
When Friend node send model message to LPNs, 
firstly: the message encrypt use application key and application nonce(**include SEQ**).
secondly: the **same seq** should also use to encrypt in network layer.

Obviously, this association is interrupted in the new code. In other words, seq will be inconsistent at different layers. Eventually leading to decryption failure.

jhedberg, 2019-09-19:
Does #18745 address this?

menglingao1997, 2019-09-19:
Not only has this problem not been solved, it is precisely because of this pattern that this has happened.

menglingao1997, 2019-09-19:
[image]

trond-snekvik, 2019-09-19:
Looks like I screwed up. I'll take a look and see if this can be fixed properly without too big a change. Otherwise, we should probably revert #18745, as it just replaces one corner case with a much bigger one.

menglingao1997, 2019-09-19:
#18745 Can effectively process messages sent by other nodes to LPNs, However The friend self have  issure, and segment message may also present this problem, but due to present `seg_zero`  filed, so segment message will compute correctly `seq` even not Inconsistent between different layer.

menglingao1997, 2019-09-19:
but, when `segment` message `seq` is `8192` higher than `seqauth`, will be incorrect.

menglingao1997, 2019-09-25:
I'll take a stab at this :)

jhedberg, 2019-09-25:
> I'll take a stab at this :)

Hold on. If I understood right @trond-snekvik already has some code for this, but perhaps didn't yet open a PR. @trond-snekvik can you confirm?

trond-snekvik, 2019-09-25:
Hey, yeah I've got a branch in progress, but I've been held up by some other issues the last few days

PR #19475 (2019-10-04): Bluetooth: Mesh: Friend enc failure
Re-encrypts single-segment application messages when the network seqnum has changed, to avoid encrypting messages with different seqnums in network and transport. This operation is only required for unsegmented messages, as segmented messages don't need to use the same seqnum in network.

Fixes #19265.
commit a7cf776d (2019-10-04): Bluetooth: Mesh: Ensure seqnum match in app/net

Re-encrypts single-segment application messages when the network seqnum
has changed, to avoid encrypting messages with different seqnums in
network and transport. This operation is only required for unsegmented
messages, as segmented messages don't need to use the same seqnum in
network.

Reinstates the special adv data for friend messages to store the app key
index.

Fixes #19265.
closed 2019-10-04 by jhedberg