zephyrproject-rtos/zephyr #19284: Service Changed indication not being sent in some cases

fnde-ot, 2019-09-19:
**Describe the bug**
Service Changed data is not being persisted so in some cases a subscribed peer will not receive an indication if the GATT database changes due to a device firmware update.

**To Reproduce**
1. bonded device is disconnected
2. reboot with a new firmware containing GATT database changes
3. stored database hash does not match calculated database hash
4. new database hash is stored
5. reboot
6. stored database hash matches calculated database hash
7. bonded device reconnects and does not receive any SC Indication

This only affects older devices that do not support robust caching feature, or when CONFIG_BT_GATT_CACHING=n.

**Expected behavior**
Reconnected peer should always receive an SC indication if the GATT database has changed since its last connection.

**Describe the solution you'd like**
- Store SC data in a new structure similar to existing CF data:
struct gatt_sc_cfg {
	u8_t id;
	bt_addr_le_t peer;
	u8_t data[4];
};

#define SC_CFG_MAX (CONFIG_BT_MAX_PAIRED + CONFIG_BT_MAX_CONN)
static struct gatt_sc_cfg sc_cfg[SC_CFG_MAX] = {};

- Create a new "bt/sc/..." setting to store that SC data (similar to "bt/cf/..." setting)

- Persist SC data when changes occur (db hash changed after reboot, reg/unreg gatt service)

**Describe alternatives you've considered**
We could merge SC and CF data in a single struct / setting, but setting format would then depend on CONFIG_BT_GATT_CACHING parameter which could make things tricky.

aescolar, 2019-09-19:
@fnde-ot Do you want to assign this to yourself of @Vudentz ?

fnde-ot, 2019-09-20:
I can have a go at it unless @Vudentz prefers to take it.

PR #19501 (2019-10-15): Bluetooth: GATT: Persist Service Changed data
Fixes #19284.

Add support for persisted Service Changed data, to fix the case of a
paired device not reconnecting before a reboot and thus not receiving
SC indication. It also enables support for GATT database being changed
during a firmware update.

- Add a BT_GATT_SERVICE_CHANGED config option to enable SC support. Both
BT_GATT_CACHING and BT_GATT_DYNAMIC_DB depend on this new parameter.

- Move Service Changed data outside of the CCC struct and make it
persistent by adding support for a "bt/sc/..." setting.
closed 2019-10-15 by carlescufi