zephyrproject-rtos/zephyr #19299: kernel/spinlock: A SMP race condition in SPIN_VALIDATE

cwshu, 2019-09-20:
**Describe the bug**
This bug is same as #14807 but it is closed, So I open a new issue to report it. (If this isn't the right way to reopen an issue, then would you kindly tell me how to do it?)

When both SPIN_VALIDATE and SMP(cpu_num = 2) on, z_spin_lock_valid() occasionally generates assertion of "recursive spinlock" when I run samples/synchronization or tests/kernel/smp. I find there is a race condition in z_spin_lock_valid(), so it will trigger "recursive spinlock" assertion despite no recursive spinlock occurs.

z_spin_lock_valid() reads shared variable "l->thread_cpu" twice. If CPU1 set "l->thread_cpu" to 0 (by z_spin_unlock_valid()) between two read accesses from z_spin_lock_valid() at CPU0, then CPU0 will pass two checkings and find that it hold the recursive spinlock. The first checking is passed because the lock is hold by CPU1's thread, and the second checking is passed because setting "l->thread_cpu" to 0 lets CPU0 find the lock is holding by CPU0.

**To Reproduce**
I run the program on the custom platform, but I think that qemu_x86_64 can also reproduce this bug.

Steps to reproduce the behavior (use samples/synchronization):
1. mkdir build; cd build
2. cmake -DBOARD=qemu_x86_64
3. make menuconfig   # enable DEBUG and ASSERT
4. make
5. run program

**Note of bug reproduce**
This race condition occurs occasionally. Adding some delay between two read accesses in z_spin_lock_valid() make it happen easily.

```
bool z_spin_lock_valid(struct k_spinlock *l)
{
    if (l->thread_cpu) {
        k_busy_wait(1);
        if ((l->thread_cpu & 3) == _current_cpu->id) {
            return false;
        }
    }
    return true;
}
```

**Expected behavior**
Not print assertion fail of recursive spinlock.

**Screenshots or console output**
***** Booting Zephyr OS 1.14.0 *****
...
ASSERTION FAIL [z_spin_lock_valid(l)] @ $ZEPHYR_BASE/include/spinlock.h:79
        Recursive spinlock
@ $ZEPHYR_BASE/lib/os/assert.c:30:
***** Kernel Panic! *****
Current thread ID = 0x0000a16c
Faulting instruction address = 0xdeadbaad
  ra: 0xdeadbaad  gp: 0xdeadbaad  tp: 0xdeadbaad  t0: 0xdeadbaad
  t1: 0xdeadbaad  t2: 0xdeadbaad  t3: 0xdeadbaad  t4: 0xdeadbaad
  t5: 0xdeadbaad  t6: 0xdeadbaad  a0: 0xdeadbaad  a1: 0xdeadbaad
  a2: 0xdeadbaad  a3: 0xdeadbaad  a4: 0xdeadbaad  a5: 0xdeadbaad
  a6: 0xdeadbaad  a7: 0xdeadbaad

**Environment**
 - OS: Linux
 - Zephyr v1.14.0

aescolar, 2019-09-20:
CC @andrewboie @andyross

andyross, 2019-09-20:
Ah, indeed.  Yeah, I see it.   Just to be clear: you can only see this failure on CPU0, because only there is the zeroed-out thread_cpu value ambiguous with the CPU ID?  I don't see a patch yet, can you confirm that a fix like:

```c
--- a/kernel/thread.c
+++ b/kernel/thread.c
@@ -808,8 +808,10 @@ FUNC_NORETURN void k_thread_user_mode_enter(k_thread_entry_t entry,
 #ifdef SPIN_VALIDATE
 bool z_spin_lock_valid(struct k_spinlock *l)
 {
-       if (l->thread_cpu) {
-               if ((l->thread_cpu & 3) == _current_cpu->id) {
+       uintptr_t thread_cpu = l->thread_cpu;
+
+       if (thread_cpu) {
+               if ((thread_cpu & 3) == _current_cpu->id) {
                        return false;
                }
        }
```

...resolves the issue?  If so, do you want to submit it or should I?

andyross, 2019-09-20:
Just to repeat my understanding to make sure it's the same as yours:

1. CPU1 holds the lock, so thread_cpu is non-zero
2. CPU0 tries to acquire the lock, and checks thread_cpu (which is non-zero)
3. CPU1 releases the lock, making thread_cpu zero
4. CPU0 checks the CPU field of the value and finds a zero there (because the whole field is non-zero), which matches its own CPU ID, which makes it think it already owned the lock, which is an error.

cwshu, 2019-09-21:
> Ah, indeed. Yeah, I see it. Just to be clear: you can only see this failure on CPU0, because only there is the zeroed-out thread_cpu value ambiguous with the CPU ID?

Yes, this failure is only on CPU0.

> I don't see a patch yet, can you confirm that a fix like:
> (pass)
> ...resolves the issue? If so, do you want to submit it or should I?

Yes, this bugfix can resolve the race condition.
I can run it without assertion when adding some different delays.

If you are ok, I want to submit this patch.

> Just to repeat my understanding to make sure it's the same as yours:
>
> 1. CPU1 holds the lock, so thread_cpu is non-zero
> 2. CPU0 tries to acquire the lock, and checks thread_cpu (which is non-zero) [...]

Yes, it is same as my understanding.

PR #19400 (2019-09-26): kernel/spinlock: Fix a SMP race condition of SPIN_VALIDATE
z_spin_lock_valid() reads shared variable twice to do two checkings. If
this variable is modified by other CPU between two read accesses, the
checking value is inconsistent. This inconsistency causes the error
that CPU0 can pass the checking when it doesn't hold spinlock because
zeroed-out thread_cpu value is ambiguous with the CPU0 ID.

Fix the inconsistency by only reading shared variable once and using
local variable value to do two checkings.

Fixes #19299.
commit e124670f (2019-09-26): kernel/spinlock: Fix a SMP race condition of SPIN_VALIDATE

z_spin_lock_valid() reads shared variable twice to do two checkings. If
this variable is modified by other CPU between two read accesses, the
checking value is inconsistent. This inconsistency causes the error
that CPU0 can pass the checking when it doesn't hold spinlock because
zeroed-out thread_cpu value is ambiguous with the CPU0 ID.

Fix the inconsistency by only reading shared variable once and using
local variable value to do two checkings.

Fixes #19299.
closed 2019-09-26 by nashif