zephyrproject-rtos/zephyr #19307: _interrupt_stack is defined in the kernel, but declared in arch headers

gnuless, 2019-09-20:
`kernel/init.c` defines `_interrupt_stack` (and on SMP systems, additional per-CPU interrupt stacks), but no kernel header file exports a declaration.  As a result, most (if not all) arch trees redeclare it themselves (and `kernel/smp.c` redeclares the additional stacks).  These declarations should all be consolidated, probably in `kernel_internal.h`.

ioannisg, 2019-09-22:
Got this. I think, if we go for it, we could try to address #16210, if possible.

ioannisg, 2019-09-22:
FYI @andrewboie

ioannisg, 2019-09-24:
BTW,  @gnuless is it possible to have a Zephyr build without an interrupt stack?

andrewboie, 2019-09-24:
> BTW, @gnuless is it possible to have a Zephyr build without an interrupt stack?

Why would you want to do this?

closed 2019-10-18 by gnuless