zephyrproject-rtos/zephyr #19325: networking: using external buffers for tx and rx

ehudmarvell, 2019-09-23:
Hi,

I am using hw with buffer management both rx&tx.

Therefore, I need to use external buffers for rx & tx.

Currently, as I understand the zephyr support as following:
RX:
on rx it seem I can use "net_buf_alloc_with_data" but no callback registration exist while I call data_unref.

TX:
As I understand,  I can't use external buffers here, just using fixed or var buffers which allocated static.

I will glad to here that I missing available implementation, but if not,
can you suggest what do you think?

 
Thanks,
Ehud

jukkar, 2019-09-23:
> I am using hw with buffer management both rx&tx.

Can you elaborate more what you mean by that? Do you mean offloaded network stack?

ehudmarvell, 2019-09-25:
it is king of offload network, on rx the hw fill buffers which I fill before and I can move it to os without any copy. On tx, after the hw sending the buffer it return it back to its(hw bm) pool without any SW actions

jukkar, 2019-09-25:
This sounds similar as what `drivers/ethernet/eth_sam_gmac.c` is doing, please check if you could do similar things in your driver.

ehudmarvell, 2019-09-26:
Hi Jukkar,

Thank you for your answer but the driver didn't seem good enough for my needs.
for example:
1. the RX buffer are allocated one on init and I need the option change then on run time.
2. As I understand this driver, the tx buffers also given by the os and I need to control thus buffers.

Thanks,
Ehud

carlescufi, 2023-11-15:
@ehudmarvell is this still an issue or should we close it? This seems a bit specific, perhaps we need a zero-copy support in the networking subsystem issue instead?

closed 2025-03-05 by carlescufi