zephyrproject-rtos/zephyr #19326: Zephyr Incubator Module

Laczen, 2019-09-23:
## Introduction

As zephyr is evolving there is a continuous struggle in getting new features approved and maintaining the required code quality. This proposal  discusses a method that allows developers to propose and submit new zephyr extensions that can be tested by users while maintaining the required code quality.

### Problem description

When developers are proposing a new extension to zephyr it is required to do this directly to the zephyr tree. This way the extension is reviewed and is added to the zephyr tree when development is of sufficient quality for inclusion. During the development there is only limited user feedback as the extension is not easily available for other users. As soon as the module is available in the zephyr tree feedback from users is used to further improve the extension. Due to the review process it can take a long time before a new extension is available to users.

### Proposed change

The (recent) addition of zephyr modules can be used to transform the process of getting new extensions in the zephyr tree. I propose to create a zephyr incubator module (zephyr_incubator) and repository. This repository can be used by any user to submit (and share) new zephyr extensions as a zephyr module. The review process in this repository is relaxed. 

In the zephyr_incubator modules can grow and improve quality while getting user feedback. Every 3 months (or after request from the author) the dev-review meeting is used to determine what modules are ready to be integrated in the main zephyr tree and the current review process is started. The integration can be very easy when the zephyr extension has been developed as a zephyr module. 

### Proposed change (Detailed)

Setting up a zephyr incubator repository is simple. It can be done like any other zephyr module. To make it easy for developers to add their own module a "umbrella" `CMakeLists.txt` and a "umbrella" `Kconfig` is added at the top level. A zephyr folder with a `module.yml` is added to point to the umbrella `CMakeLists.txt` and `Kconfig.txt`. An example of how this can be done can be seen at https://github.com/Laczen/zephyrmodules.git. To complete the incubator module I would add a sample folder that contains a example `CMakeLists.txt` and `Kconfig`. New developers can start from this sample to create their own module. To ensure documentation and testing is done from the start an example doc and test folder can be added to the sample.

The creation of a new module is done by copying and renaming the sample folder to an appropriate name and adding it to the umbrella `CMakeLists.txt` and 'Kconfig`. As soon as the module is added to zephyr incubator (which could be a simple PR without review) other users can use the new module and help the author in getting it up to the required code quality.

jfischer-phytec-iot, 2019-09-23:
> During the development there is only limited user feedback as the extension is not easily available for other users. As soon as the module is available in the zephyr tree feedback from users is used [...]

I do not understand where the problem lies, why is it not available? If an extension is a PR then a user can check it out and test it.

Laczen, 2019-09-23:
Hi @jfischer-phytec-iot, thanks for the comment. The extensions are is not easily available, you need to search the PR list to find it.

galak, 2019-09-26:
Changing to a TSC discuss as that the proper forum for this at this point.

@MaureenHelm can you try and get with @Laczen to get on the TSC agenda.

closed 2023-04-27 by Laczen