zephyrproject-rtos/zephyr #19328: Logger could block in thread at certain log message pool usage

nordic-krch, 2019-09-23:
**Is your enhancement proposal related to a problem? Please describe.**
#16194 introduces new feature to the logger which will allow blocking in threads when there is log message pool is empty (instead of dropping or overwritting the message). It happens when pool is empty so contexts which cannot block will start dropping messages at that point.

**Describe the solution you'd like**
Instead of blocking when pool is empty, it could block at certain availability level which would allow non-blockable context to use the logger (and not drop or overwrite messages). Currently, straight forward approach is used (timeout argument in mem slab alloc). Configurable threshold would require additional implementation.

nashif, 2021-05-28:
@nordic-krch  can this be closed?

nashif, 2021-06-23:
@nordic-krch  ping

nordic-krch, 2021-06-23:
It can be reopened when there is some interest.

closed 2021-06-23 by nordic-krch