zephyrproject-rtos/zephyr #19340: ARM: Cortex-M: Stack Overflows when building with NO_OPTIMIZATIONS

ioannisg, 2019-09-24:
**Describe the bug**
We see stack overflows when building tests and samples with `CONFIG_NO_OPTIMIZATIONS=y` various ARM platforms. [Besides, of course, tests or samples that do not build at all with this flag, due to resource limitations, which is, perhaps, OK.]

**To Reproduce**
Just build tests with `CONFIG_NO_OPTIMIZATIONS=y` and `CONFIG_HW_STACK_PROTECTION=y` on MPU-enabled ARM boards, and most likely will be hit by Stack Overflow.

**Expected behavior**
I would like to be able to execute the samples and tests in ARM platforms with CONFIG_NO_OPTIMIZATIONS=y, provided that the flash and ram are sufficient.

**Impact**
Minimal. But perhaps we would like to provide a core thread stack configuration that is sufficient for the case of no-optimization.

**Additional context**
- It could be as simple as increasing MAIN and IDLE (and possibly ZTEST) thread stack sizes when building with `CONFIG_NO_OPTIMIZATIONS=y`.

- Or we document in the NO_OPTIMIZATIONS option that enabling this will, possibly, require a different stack size configuration.

@andrewboie @galak is this something we would like to do? Has this issue already been brought up and classified as "no-action"?

andrewboie, 2019-09-24:
> @andrewboie @galak is this something we would like to do? Has this issue already been brought up and classified as "no-action"?

We need to fix this because code coverage requires optimizations disabled.

First step is to identify what tests are failing, we can use results for mps2_an385 to determine this

SebastianBoe, 2019-09-25:
> Or we document in the NO_OPTIMIZATIONS option that enabling this will, possibly, require a different stack size configuration.

I think this is reasonable.

> It could be as simple as increasing MAIN and IDLE (and possibly ZTEST) thread stack sizes when building with CONFIG_NO_OPTIMIZATIONS=y.

We are already doing this for various other stacks, so doing so for these stacks as well would follow prior-art and sounds reasonable.

There is no known clean way to solve this problem in the general case.

ioannisg, 2019-09-26:
@andrewboie @SebastianBoe thanks for the feedback.
To summarize:

> First step is to identify what tests are failing, we can use results for mps2_an385 to determine this

Right. I am going to create a list of failing tests, and the corresponding HW platforms. And I am going to try determining the minimum IDLE, MAIN, and ZTEST stack size that fits all ARM platforms.

>Or we document in the NO_OPTIMIZATIONS option that enabling this will, possibly, require a different stack size configuration.

I see that this is fine for application developers, but won't make the tests pass in the tree with NO_OPTIMIZATIONS

>It could be as simple as increasing MAIN and IDLE (and possibly ZTEST) thread stack sizes when building with CONFIG_NO_OPTIMIZATIONS=y.

OK, so let's just do that.

> We need to fix this because code coverage requires optimizations disabled.

@andrewboie Currently, we only enable COVERAGE on the coverage-specific x86 Qemu, correct?

Then: do you suggest that all testing with COVERAGE=y need to select NO_OPTIMIZATIONS=y? We do not seem to do this.

Also: do you see value of including test-case variants in selected tests with NO_OPTIMIZATIONS=y? Currently, I am doing this for tests/arch/arm/arm_thread_swap only.

andrewboie, 2020-10-07:
you could set a large-ish CONFIG_TEST_EXTRA_STACKSIZE for the ARM qemu targets which would probably allow this to work for tests in the emulator.

I don't have bandwidth to work on this one at the moment.

emillindq, 2020-11-09:
To bump this a bit, I'd like to mention we encountered this issue in our project when pulling latest Zephyr (2.4.0 rc1 -> 2.4.99). We encountered this when building our app. It seems like it only occurs when compiling in extra modules. We use `CONFIG_NO_OPTIMIZATIONS` (it seems like `CONFIG_DEBUG` is not good enough), as well as `CONFIG_ASSERT` and `CONFIG_LOG`. We use stm32h747i_disco_m7 board.

Add those configs to hello_world sample app and you'll get MPU FAULT. The issue resolves when increasing idle task stack size. Default is 320, if we increase to 340 issue is resolved.

carlescufi, 2021-02-09:
API meeting 9th Feb.

@ioannisg this seems more effort than it's worth, because it is expected that code that is less optimized will consume more stack. Unless you have a good reason to keep this open, the consensus today seems to be to close it.

ioannisg, 2021-02-10:
> API meeting 9th Feb.
> 
> @ioannisg this seems more effort than it's worth, because it is expected that code that is less optimized will consume more stack. Unless you have a good reason to keep this open, the consensus today seems to be to close it.

@carlescufi honestly, one of the (main) reasons to keep this open would be to close any duplicate tickets community members might want to open when they observe relevant issues; better to have one ticket that would cover all such failure reports.

I don't have strong arguments for keeping the ticket open, though. One option would be to close it, if we are already documenting (properly) that building without optimizations may lead to increased thread stack requirements. I'll check this out.

closed 2021-04-27 by github-actions[bot]