zephyrproject-rtos/zephyr #19347: BL654 USB dongle not found after flashing

kristiankajak, 2019-09-24:
Dear Zephyr community, hope you can help me out with this one.

**Describe the bug**
The Laird BL654 451-00004 USB dongle disappears from the USB devices list after flashing the zephyr .hex file onto it. The LED turns off and there is no indication of whether it is still working or not. Furthermore, I can't attach minicom to it for debugging purposes due to not being able to find the device.

I have tried flashing _hci_uart_ and _beacon_ and _hello_world_ examples on for _nrf52840_pca10059_ and _bl654_dvk_, using the official 451-00004 flashing [commands stated by Laird](https://connectivity-staging.s3.us-east-2.amazonaws.com/2019-07/CS-QSG-BL654-USB%20_Nordic_Zephyr_v1_0.pdf). Furthermore, I have tried flashing it with both Zephyr 1.14 and 2.0 releases.

Previously I have compiled and flashed for the nrf52_pca10040 board without problems.

**To Reproduce**
Steps to reproduce the behavior:
1.  _cd samples/bluetooth/hci_uart_
2.  _west build -p -b nrf52840_pca10059_
3. _nrfutil pkg generate --hw-version 52 --application build/zephyr/zephyr.hex --application-version-version 1 --sd-req 0x00 out.zip_
4.  _nrfutil dfu usb-serial -pkg out.zip -p /dev/ttyACM0_
5. "_Device programmed_", but the dongle disappears under _lsusb_

**Expected behavior**
Device is programmed and is still accessable from _/dev/ttyACM0_

**Impact**
showstopper

**Environment (please complete the following information):**
 - OS: Linux Mint 19.2 Tina, Linux kernel 4.15.0-54-generic
 - Zephyr 1.14 (cebe115) and Zephyr 2.0  (7c0ec43)
 - nrfutil version 5.2.0
 - West v0.6.2
 - nrfjprog version: 10.3.0
 - JLinkARM.dll version: 6.50b

**Additional context**
I suspect the problem is with the board selection when executing the west command but then again, I can't see any better board candidates either.

jfischer-phytec-iot, 2019-09-24:
You flashed an example without USB functionality and expect USB host to enumerate the device? Please try `samples/subsys/usb/console`.

kristiankajak, 2019-09-24:
Thank you, this solved it. Well, the examples worked fine on the _nrf52_pca10040_ board. However, I did not realize that the _nrf52_pca10040_ was able to retain the USB connection because of the on-board SEGGER J-Link which the USB dongle doesn't have.

jfischer-phytec-iot, 2019-09-24:
>However, I did not realize that the nrf52_pca10040 was able to retain the USB connection because of the on-board SEGGER J-Link which the USB dongle doesn't have.

No, it is nRF52832 UART connected over J-Link's CDC-ACM function
[nRF52832] <-uart-> [Debug Probe's CDC-ACM] <-usb-> [Host]

carlescufi, 2019-09-25:
Resolved.

closed 2019-09-25 by carlescufi