zephyrproject-rtos/zephyr #19356: LwM2M sample reorganization: split out LwM2M source into object-based .c files

mike-scott, 2019-09-24:
**Is your enhancement proposal related to a problem? Please describe.**
The LwM2M sample has become fairly complex over the last 2 years.  I propose to break out each LwM2M object / configuration into it's own source file.

**Describe the solution you'd like**
Each object-based .c file could be located under the `src/lwm2m/` directory.  This way new users of the code can easily locate things they are interested in like the `lwm2m_button.c` file or `lwm2m_firmware.c`.

**Describe alternatives you've considered**
The sample is only going to get more complex as I add support for different network type, so this is needed prior to those changes.

mike-scott, 2019-09-24:
@endian-benjamin @retfie FYI

dleach02, 2019-12-10:
Removed the 2.1 milestone.

PR #50528 (2022-10-11): samples: net: lwm2m_client: Split sensor modules out from main
Each module deserves their own file so we don't clutter the main application.

Fixes #19356
closed 2022-10-11 by carlescufi