zephyrproject-rtos/zephyr #19361: BLE Scan fails to start when running in parallel with BLE mesh

stephenmasih, 2019-09-25:
**Describe the bug**
When I try to start the scan before mesh initialising followed by self-provisioning and configuring. It fails with error code -11.
When I first successfully initialise, provision and configure the mesh. and then call bt_le_scan_start(). It fails with error code -69. But I am able to publish and subscribe to the mesh messages. 

**To Reproduce**
Steps to reproduce the behaviour:
1. mkdir build; cd build
2. cmake -GNinja -DBOARD=nrf52840_pca10056 ..
3. ninja
4. See error

**Expected behaviour**
BLE mesh works in broadcaster and scanning topologies. I should be able to scan all nearby beacon packets and parallel stay involved in the mesh.

**Screenshots or console output**
[image]
[image]

jhedberg, 2019-09-25:
Unfortunately this is currently by design. The Mesh stack essentially "owns" the advertising and scanning functionality and any attempts to make direct calls to `bt_le_adv_start()` or `bt_le_scan_start()` will result in errors or messing up the state of the mesh stack. We do have plans to fix this, but that involves having a controller that either supports Advertising Extensions, or Mesh HCI vendor extensions.

stephenmasih, 2019-09-25:
I see, can you tell if there are any possible ways because the mesh follows the observer capabilities. It picks up packets and does the work accordingly. Can I filter out beacon packet of a particular type somewhere in the zephyr code?

jhedberg, 2019-09-25:
@stephanosio currently you'd need to manually edit the `bt_mesh_scan_cb() function in `subsys/bluetooth/mesh/adv.c` to add support for non-mesh advertising packets.

I would like to introduce a more flexible scan API which would allow having multiple callbacks, perhaps even per advertising data type, however I put this work on hold when I realised we'd probably need to property take into account Advertising Extensions as well. My worry was that since AE requires new APIs we'd end up having three different ways of catching advertisements. Anyway, the simple idea was to just have a linked list of callbacks, using something like this:
```
--- a/include/bluetooth/bluetooth.h
+++ b/include/bluetooth/bluetooth.h
@@ -454,6 +454,20 @@ int bt_le_scan_start(const struct bt_le_scan_param *param, bt_le_scan_cb_t cb);
  */
 int bt_le_scan_stop(void);
 
+typedef void bt_scan_data_func_t(struct bt_scan_data_cb *cb,
+                                const bt_addr_le_t *addr, s8_t rssi,
+                                u8_t adv_type, struct net_buf_simple *buf);
+
+struct bt_scan_data_cb {
+       sys_snode_t         node;
+       bt_scan_data_func_t func;
+       u8_t                data_type;
+       u8_t                adv_types;
+};
+
+void bt_le_scan_data_cb_add(struct bt_scan_data_cb *cb);
+
+void bt_le_scan_data_cb_del(struct bt_scan_data_cb *cb);
```

If we can do something like that which also fits nicely together with AE APIs then that would be ideal from my perspective. @joerchan FYI.

stephenmasih, 2019-09-25:
Thank you @jhedberg FYI. Let me try editing the bt_mesh_scan_cb() in adv.c . Can you please tell me if it is possible to work with mesh beacons. As I see,
case BT_DATA_MESH_BEACON:
			bt_mesh_beacon_recv(buf);
			break;
What kind of beacon packet type does it pick up?

jhedberg, 2019-09-25:
> What kind of beacon packet type does it pick up?

@stephenmasih see section "3.9 Mesh beacons" in the Mesh Profile Specification.

stephenmasih, 2019-10-02:
I added a callback function as you mentioned @jhedberg. My callback is getting triggered but my scanning stops after some time the mesh initialisation. I mean even my mesh data packets are dropped. My callback is also neither triggered anymore. I seems like it crashed. Can you explain to me the probable reason?

jhedberg, 2019-10-02:
@stephenmasih I wouldn't dare to claim a probable reason, but some common reasons can be stack overflows and controller asserts. I'd recommend enabling various logging/debugging Kconfig options to see if you can get some error outputted to the console.

jhedberg, 2019-10-02:
@stephenmasih if you have gdb access it would also help to use that to investigate in what state the system is

joerchan, 2020-01-07:
If we can do something like that which also fits nicely together with AE APIs then that would be ideal from my perspective. joerchan FYI.

@jhedberg I think this is covered by @trond-snekvik currently open PR.

@stephenmasih Any updates on this? We have a similar  bug-report for the advertiser, maybe it is the same type of issue for the scanner? It could be affecting the scanner disable call.
https://github.com/zephyrproject-rtos/zephyr/issues/21498

carlescufi, 2020-02-14:
@jhedberg @trond-snekvik can this be closed?

jhedberg, 2020-02-14:
> @jhedberg @trond-snekvik can this be closed?

I'd say so. @stephenmasih can reopen it if/when he's done with further digging and concluded that there's still something to fix in the stack.

closed 2020-02-14 by jhedberg