zephyrproject-rtos/zephyr #19376: Build on a ARM host

codingspirit, 2019-09-25:
Dear zephyr community, hope you can help me on this. Stuck here for a while and any suggestions will be appreciated. Thanks a lot.

**Describe the bug**
I'm trying to setup zephyr build environment on raspberry pi, so the host will be raspberry pi and the target will be stm32f4 MCU. It seems that the SDK binaries provided by zephyr is for host x86-64 only. Is there any other way to compile zephyr for cortex-m mcu on an ARM host with/without zephyr SDK? 

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b reel_board samples/hello_world
2. cmake report error when trying to run i586-zephyr-elf-gcc because it's x86-64 binary

**Expected behavior**
Successfully built hello_world on pi.

**Impact**

**Screenshots or console output**

**Environment (please complete the following information):**
 - OS: (Linux on PI)
 - Toolchain (Zephyr SDK)

**Additional context**

galak, 2019-09-25:
If there's arm embedded build of GCC with the distro you are running on the RPI than you could try the instructions for using a generic toolchain here:

https://docs.zephyrproject.org/latest/getting_started/toolchain_other_x_compilers.html

codingspirit, 2019-09-25:
@galak  Thanks, I will try that

closed 2019-09-25 by codingspirit

oberien, 2020-07-06:
@codingspirit Were you able to build zephyr on an RPi? If yes, which environment variables did you need to set to which values? I'm trying `ZEPHYR_TOOLCHAIN_VARIANT="cross-compile"` and `CROSS_COMPILE="/usr/bin"`, but I get the error that Zephyr was unable to find the toolchain, showing those two values.

trunifom, 2020-07-23:
Have you solved the problem?

I had the same issue.

oberien, 2020-07-23:
I just installed `arm-none-eabi-gcc` on the RPi by following the instructions on https://github.com/vanbwodonk/gcc-arm-embedded-build-armhf (tested on commit 8219275266e81ac9ab205ffb1d82db55f32d50fe). With that toolchain, compilation works perfectly normal and fine.