zephyrproject-rtos/zephyr #19380: Potential bugs re. 'static inline' and static variables

ulfalizer, 2019-09-25:
Prompted by some discussion on Slack, haven't looked for problems.

If you put

```c
static inline void f(void)
{
	static int x;

	...
}
```

in a header, you'll get a separate `x` variable for each translation unit that includes it. It works just like a regular `static` function there. The `inline` makes no difference.

@tautologyclub might've found some stuff that assumes `x` is shared.

tautologyclub, 2019-09-25:
Yeah apart from `coap_next_id()` from `coap.h` (fix incoming in https://github.com/zephyrproject-rtos/zephyr/pull/19217), I found one instance that looks slightly dubious in `subsys/net/ip/net_private.h`: 

```
static inline char *net_sprint_ll_addr(const u8_t *ll, u8_t ll_len)
{
	static char buf[sizeof("xx:xx:xx:xx:xx:xx:xx:xx")];

	return net_sprint_ll_addr_buf(ll, ll_len, (char *)buf, sizeof(buf));
}
```
Not sure it will cause any bugs, but it certainly looks shady. Note that in the case of `coap_next_id()`, it DID cause very serious bugs.

Note: my methodology for finding them is probably not perfect (basically a chain of twenty grep/cat pipes). Coccinelle might be better?

marc-hb, 2019-09-25:
I did some experimentation and disassembly with minimal examples and gcc (GCC) 8.3.1 20190223 (Red Hat 8.3.1-2). With this compiler I verified that every translation gets its own, private copy of `static int x` as expected. If that's what the author intended that I wouldn't necessarily call it "shady". Missing a comment maybe.

The bad thing about the ancient "translation units" concept is the lack of LTO and other fancy stuff. The good thing about the ancient "translation units" concept is its simplicity: the compiler treats `.h` and `.c` code the same.

> The inline makes no difference.

I also verified this.  *For these toy examples* `inline` makes absolutely zero difference. The compiled code is exactly the same with or without '`inline`. gcc does inline that simple function with -O1 and above, otherwise it doesn't; the `inline` specifier doesn't matter.

> Coccinelle might be better?

Maybe - that or some LLVM plugin/ad-hoc script?

marc-hb, 2019-09-25:
> With this compiler I verified that every translation gets its own, private copy of static int x as expected. If that's what the author intended that I wouldn't necessarily call it "shady". Missing a comment maybe.

If that's not what the author intended and the author was trying to share a global across translation units, then I wonder how that passed any testing (or code review).

@tautologyclub if you go back in time before your PR #19217 , can you find any meaningful test actually using this code and passing?

tautologyclub, 2019-09-25:
I don't think obviously inlinable functions like 'static int x; return x++' really proves anything, but I also don't believe gcc does anything but the expected (ie the variable probably always becomes translation-unit global). But neither the coap example nor the one I pasted above seems like good use cases for such statics. The question is if there IS a good use case for them - to me they seem to be just as bad (or worse) style wise as bracketless ifs, trigraphs, weird switch case/goto hacks, long jumps etc - just begging for bugs for no practical gain.

marc-hb, 2019-09-25:
> I don't think obviously inlinable functions like 'static int x; return x++' really proves anything

It doesn't prove anything wrt to optimizations and speed, however I found it interesting wrt to this particular issue  and the general concept that `inline` never makes any *functional difference*. `static` is all what matters here.

> The question is if there IS a good use case for them -

I just posted this in #19217, would have been better here sorry:

> I could think of some corner case(s) where you want to avoid the stack growing too much and you know for sure there won't be any concurrent use but... yeah, that would deserve a big fat comment at the very least.

BTW neither `gcc -Wall -Wextra`  8.3.1 20190223 nor `clang -Wall -Wextra` 7.0.1-6.fc29 says anything when doing this.

jukkar, 2019-09-25:
> I found one instance that looks slightly dubious in `subsys/net/ip/net_private.h`:
>
> ```
> static inline char *net_sprint_ll_addr(const u8_t *ll, u8_t ll_len)
> {
> 	static char [...]

This indeed looks shady, but actually using static buf here is good (although it was unintentional). This func is only used in various debug prints and having static buf here makes it unlikely that various debug prints would overwrite each other. Only issue is that we are increasing memory consumption but luckily it is only happening if debug level is high.

marc-hb, 2019-09-25:
So @jukkar you're saying the code above is a trade-off between memory use and correctness where invocations from different translation units are isolated from each other but invocations from the same translation unit are not and could (harmlessly?) corrupt debug output? Just rephrasing to make sure I understood.

> but... yeah, that would deserve a big fat comment at the very least.

tautologyclub, 2019-09-25:
Sounds pretty ad hoc to me. Either you need reentrancy or you don't. What happens if I call that sprintf function twice in the same stack? Etc... Can you truly 100% vouch for correctness in all such possible situations?

tautologyclub, 2019-09-25:
@marc-hb regarding tests etc, I don't believe the coap function was rigorously tested, at least not the aspect of getting called from multiple translation units. It's such a meta thing to test though so I can't blame the author.

jukkar, 2019-09-25:
> So @jukkar you're saying the code above is a trade-off between memory use and correctness where invocations from different translation units are isolated from each other but invocations from the [...]

There was no intentional trade-offs here :) The function in question was never meant to guarantee that debug output is not corrupted. So originally the idea was to minimize memory usage by using static variable. As the function is inlined, it is actually harder to corrupt the output which is nice in this case.

SebastianBoe, 2019-09-26:
The compiler is behaving as expected, removing bug label.

marc-hb, 2019-09-26:
> The function in question was never meant to guarantee that debug output is not corrupted

I realize this function is located in a file named `something_PRIVATE.h`, yet I still think this deserve a big comment/warning in the code, something like "very sharp knife here, don't do this at home", etc.

ceolin, 2019-10-18:
my 5 cents here, we should forbidden this kind o feature. This is totally error prone, the variable is defined in each translation unit that the function is used, and the behaviour is likely not what the developer wanted, this will work as static for each translation unit but not globally.

@marc-hb  @tautologyclub is there anything gating this change ? I can easily do a coccinelle script to change / enforce it.

tautologyclub, 2019-10-19:
Well I think it definitely should be done, but I didn't get the impression that there is buyin from everyone involved. To me that sprintf looks like it's 100% begging for trouble.

andrewboie, 2019-10-19:
> my 5 cents here, we should forbidden this kind o feature. This is totally error prone, the variable is defined in each translation unit that the function is used, and the behaviour is likely not what the developer wanted, this will work as static for each translation unit but not globally.

I agree. People should not be doing this.

tautologyclub, 2019-10-22:
@ceolin first you should find the instances with coccinelle, I doubt my search was exhaustive

d3zd3z, 2019-12-13:
There is a deeper problem, that `static inline ...` isn't actually the best way to do inline functions in gcc. It usually works, except that depending on certain things, like this construct, it will end up putting code for the function in every module that uses it. This should be warnable against, by adding `-Winline`, which will warn about function that can't be inlined.

Another approach is to use `extern inline` in the header file. This will generate a link error if something cannot be inlined. It is possible to use ifdefs and have a single C file include the inline, without the `extern` which will generate code for the function.

It probably isn't worth converting to this, and better to just avoid static variables in an inline function.

closed 2020-10-04 by github-actions[bot]