zephyrproject-rtos/zephyr #19385: compilation error

WilliamGFish, 2019-09-25:
**Describe the bug**
When compiling receive following error:
```
Device tree configuration written to ../sensor_server/build/zephyr/include/generated/generated_dts_board.conf
warning: please put 'required: false' instead of 'category: optional' in 'properties: center-aligned: ...' in ../dts/bindings\pwm\nordic,nrf-pwm.yaml - 'category' will be removed
warning: please put 'required: false' instead of 'category: optional' in 'properties: center-aligned: ...' in ../dts/bindings\pwm\nordic,nrf-pwm.yaml - 'category' will be removed 
```

**To Reproduce**
Compile via west for NRF52840 DVK

**Expected behavior**
yaml file updated to remove error

**Impact**
annoyance, may become showstopper awaiting full impacts

ulfalizer, 2019-09-25:
Fix: https://github.com/zephyrproject-rtos/zephyr/pull/19387

WilliamGFish, 2019-09-25:
My error is with the NRF52 : /dts/bindings\pwm\nordic,nrf-pwm.yaml

I'm guesing that every Soc will require changes and is a much larger problem. Hence will need a review of all yaml files.

    _center-aligned:
      type: boolean
      description: Use center-aligned (up and down) counter mode
      category: optional_

PR #19387 (2019-09-26): dts: bindings: Remove deprecated syntax for ESP32 GPIO and NRF PWM 
Use the new 'compatible:', 'include:', and 'required:' keys, and clean
it up like other bindings.

Shorten the 'description:' text, because it appears in the output as a
comment above the generated macros, and it looks neater.

Fixes: #19385
commit d8705656 (2019-09-26): dts: bindings: Remove deprecated syntax for ESP32 GPIO and NRF PWM

Use the new 'compatible:', 'include:', and 'required:' keys, and clean
it up like other bindings.

Shorten the 'description:' text, because it appears in the output as a
comment above the generated macros, and it looks neater.

Fixes: #19385
closed 2019-09-26 by galak