zephyrproject-rtos/zephyr #19398: net: ENC28J60 driver does not respond to ping

ngiriprasad, 2019-09-26:
**Describe the bug**
I have interfaced "ENC28J60" module to the board "PCA10056". Flashed "Echo" application after making necessary configurations in the "menuconfig"(Also enabled ICMP). I have tried to ping the board from my host PC, on which Ubuntu is running. It is observed that board, is not responding to the ping requests. Captured this in "wireshark". Attached this wireshark capture as "wireshark_log_arp_enabled.zip", please go through.
As a workaround, I have made few changes realted to ARP in the file "zephyr/subsys/net/l2/ethernet/ethernet.c". Please find those changes in the attachment "ethernet_file_changes.png". After these changes, I am able to ping the board. I do not understand, why I am unable to ping the board without these changes.

**To Reproduce**
Covered under the section "Describe the bug" .

**Expected behavior**
Should be able to ping the board without making any changes to the file "zephyr/subsys/net/l2/ethernet/ethernet.c".

**Impact**
showstopper

**Environment (please complete the following information):**
 - OS: Ubuntu
 - Toolchain: Zephyr sdk 0.10.0
 - Commit SHA: bd001081fce1e2b2123f8248be9c3c938d1f470c

[wireshark_log_arp_enabled.zip](https://github.com/zephyrproject-rtos/zephyr/files/3657713/wireshark_log_arp_enabled.zip)
[image]

jfischer-phytec-iot, 2019-09-26:
Does ping6 work? Btw, a zip from "from my host PC, on which Ubuntu is running" ? and why this screenshot and not diff?

jukkar, 2019-09-26:
In the mailing list you mentioned that the ping worked from Windows linux subsystem, or did I understood something wrong?

ngiriprasad, 2019-09-28:
> Does ping6 work? Btw, a zip from "from my host PC, on which Ubuntu is running" ? and why this screenshot and not diff?

Q1) Ping6 works fine.
Q2) I mean, I am developing my application in my laptop which is running on Ubuntu. I just want to describe my working environment by that statement. Does this answer your question(Btw, a zip from "from my host PC, on which Ubuntu is running" )?
Q3) and why this screenshot and not diff?
Ans) We have cloned Zephyr into private bitbucket of our company and developing from there. That is why, I am unable to rise pull request or so for you to have diff. Please let me know if you wish to have those changes in the git hub of Zephyr. If it is required, I will do so.

ngiriprasad, 2019-09-28:
> In the mailing list you mentioned that the ping worked from Windows linux subsystem, or did I understood something wrong?

I didn't say about ping. I said about socket connection. But this issue is the root cause for that problem. If you need better clarity regarding the changes, i can create a branch on zephyr git hub, so that you will be aware of all the changes. Currently my changes are in a private bitbucket as I mentioned in the previous comment.

jukkar, 2019-10-02:
If ping6 works, then this is probably not related to device driver.
I know people who use Ubuntu to connect to device running zephyr and they have not reported these kind of issue, so I am suspecting connectivity issue in the host side.

ngiriprasad, 2019-10-03:
> If ping6 works, then this is probably not related to device driver.
> I know people who use Ubuntu to connect to device running zephyr and they have not reported these kind of issue, so I am suspecting connectivity issue in the host side.

Oh! Thanks for the information. You said, you know " people who use Ubuntu to connect to device running zephyr".  I need small clarification regarding this. Are they using "ENC28J60" or any other Ethernet chip?

ngiriprasad, 2019-10-03:
Hi,

I have raised a pull request for your reference regarding the changes I have made.  Please have a look at it and let me know if any thing goes wrong with the changes. Pull Request Number is : 19577

Thanks,
Giri

jfischer-phytec-iot, 2019-10-03:
@ngiriprasad A reference to you brunch like https://github.com/ngiriprasad/zephyr/commit/84b127bafc48963e13385290d37d878cb6beb52b is sufficient, no reason to open a PR.

ngiriprasad, 2019-10-04:
> @ngiriprasad A reference to you brunch like [ngiriprasad@84b127b](https://github.com/ngiriprasad/zephyr/commit/84b127bafc48963e13385290d37d878cb6beb52b) is sufficient, no reason to open a PR.

Oh! Okay. By the way, did you find any thing that might be causing the issue I am facing?

jukkar, 2019-10-04:
>  I need small clarification regarding this. Are they using "ENC28J60" or any other Ethernet chip?

Not with that Ethernet chip. At least the chip driver does something sane if IPv6 works.

jfischer-phytec-iot, 2019-10-04:
I can reproduce this issue on custom ENC28J60 board. Unfortunately, I have no time to dig deeper.

jukkar, 2019-10-04:
> I can reproduce this issue on custom ENC28J60 board. Unfortunately, I have no time to dig deeper.

Thanks for the info!

ngiriprasad, 2019-10-05:
> I can reproduce this issue on custom ENC28J60 board. Unfortunately, I have no time to dig deeper.

Thanks for the support @jfischer-phytec-iot . I am continuing to debug on ENC28j60. Will post any updates from my side. Please let  me know if you have any directions for me to make it faster.

ngiriprasad, 2019-10-15:
Hi Team,
While I am debugging on this issue, I found that ARP cache is not getting updated. Below is the description of the procedure I have followed and the results I got. Please go through, this may help in finding the root cause:

* I have given arping command from the host system.
* I got the arping reply from the board **PCA10056**.
* On the console of PCA10056, it is getting displayed that, **"Updating ARP cache for <host_ip_addr>"**.
* But when I am giving the command **"net arp"**, it is getting displayed as **"ARP cache is empty."**.
* Attached the log of debug prints. Please go through.

I am clueless in this behavior. Please let me know, if I should create a memory for ARP cache manually. Please help me in this regards.
[putty.log](https://github.com/zephyrproject-rtos/zephyr/files/3727914/putty.log)

jukkar, 2019-10-15:
Your changes to the ARP code basically prevent ARP to work properly, so I am not surprised that you see other issues there too.
I just wonder if the MAC address is properly setup to the enc28j60. Can you check that "net iface" properly shows the MAC address in the network interface.

ngiriprasad, 2019-10-15:
> Your changes to the ARP code basically prevent ARP to work properly, so I am not surprised that you see other issues there too.

As part of debugging, I have reverted those changes which are preventing ARP to work properly(i.e., changes in the file **"/subsys/net/l2/ethernet/ethernet.c"**). Now, ARP should work fine. Sorry for not mentioning that in the previous comment. I missed that point.

PR #19965 (2019-10-21): drivers: eth: initialize ethernet stack in enc424j600 and enc28j60
Initialize ethernet stack in drivers enc424j600 and enc28j60.

Fixes: #19398

@ngiriprasad FYI
commit 2011f737 (2019-10-21): drivers: eth: initialize ethernet stack in enc424j600 and enc28j60

Initialize ethernet stack in drivers enc424j600 and enc28j60.

Fixes: #19398
closed 2019-10-21 by jukkar