zephyrproject-rtos/zephyr #19413: Not able to scan and connect to other ble devices with HCI commands

swetha-paladugu-oyo, 2019-09-26:
Hi,

I'm using nrf52832 development board.
In which i have flashed hci_uart example code of zephyr (latest code taken from  https://github.com/zephyrproject-rtos/zephyr) which acts as an interface to send HCI commands over uart.
After flashing when i tried to send 
HCI_Reset
HCI_LE_rand
and got responses to them. After setting BD Address I tried to put the device in advertising mode by giving 
HCI_Advertising_Set_Parameters(le_scan_type : Passive, LE_scan_interval : 0x10, LE_scan_Window : 0x10, Own_Address_Type : Public Address, Scanning_Filter : 0x00)
HCI_Advertising_Enable
Advertisements are seen on nrfConnect app and is working fine.

Our usecase needs nRF to be in Master mode. For getting the devices in range I gave 
HCI_Set_Scan_Parameters
HCI_Scan_Enable
to scan the near by BLE devices, I got command complete with Success, but i didn't get scan results of the ble devices.

When i check with nRF connect app, I could see a lot of devices. Even when advertising device is kept very close, nRF is not responding with Advertisement reports.

Is there any limitation that nRF52832 with Zephyr will not work as Master mode?
Please let me know how to enable scan and connect to ble device using HCI commands.

jhedberg, 2019-09-26:
That should work just fine. Are you perhaps missing some other critical initialisation commands such as those for event mask setting (you need to set both the normal event mask to include e.g. the LE Meta event) and then the LE event mask after that. You might want to take a look at what commands the Zephyr host issues upon init. Start with the `hci_init()` function in `subsys/bluetooth/host/hci_core.c` and trace through the various functions that it calls.

carlescufi, 2019-09-26:
@swetha-paladugu-oyo please provide a full HCI trace so that we can help out further. As @jhedberg says, this is likely you failing to send commands or providing invalid parameters to them.

swetha-paladugu-oyo, 2019-09-27:
> @swetha-paladugu-oyo please provide a full HCI trace so that we can help out further. As @jhedberg says, this is likely you failing to send commands or providing invalid parameters to them.

Hi @carlescufi 
We tried to create a connection to nrf using another BT device from Cypress.
In the Cypress side we are getting LE Connection complete event and the connection seems to have been made properly.
But on the nRF side connection complete event didnt came.
But when we send some LE data from the Cypress side we could see that the LE data recieved event is coming on nRF side.
nRF always responds to commands like ReadBDADDR, LE_Scan_Enable etc but is not sending events like advertisement reports or connection complete events.
Note : We are building Zephyr with UART flow control disabled

carlescufi, 2019-09-27:
> Hi @carlescufi
> We tried to create a connection to nrf using another BT device from Cypress.
> In the Cypress side we are getting LE Connection complete event and the connection seems to have been made [...]

As mentioned earlier @swetha-paladugu-oyo, would you be able to provide a full HCI log with all commands and events exchanged between the Host and the Controller? I just don't see how we can help with good knowledge unless we see what you sent to the Controller and what it replied.

jhedberg, 2019-09-27:
@swetha-paladugu-oyo what you describe really does sound like you've failed to set the event mask. The controller might indeed be establishing a connection but it's unable to tell you because you haven't unmasked the HCI event for it.

swetha-paladugu-oyo, 2019-09-27:
> As mentioned earlier @swetha-paladugu-oyo, would you be able to provide a full HCI log with all commands and events exchanged between the Host and the Controller? I just don't see how we can help [...]

Hi @carlescufi 
[Scan_nrf.txt](https://github.com/zephyrproject-rtos/zephyr/files/3662231/Scan_nrf.txt)
[Scan_ref.txt](https://github.com/zephyrproject-rtos/zephyr/files/3662232/Scan_ref.txt)
[CreateConnection_nrf.txt](https://github.com/zephyrproject-rtos/zephyr/files/3662233/CreateConnection_nrf.txt)
[CreateConnection_ref.txt](https://github.com/zephyrproject-rtos/zephyr/files/3662234/CreateConnection_ref.txt)

We have done two experiments,
1) Creating connection from nrf to reference device
  Connection is getting created but nrf is not responding with connection complete
2) Scanning on both nrf and reference at the same time
  Scan results are coming on ref device but not on nrf
To set the event mask which HCI command should we use?
Is it LE_Set_Event_Mask

swetha-paladugu-oyo, 2019-09-27:
> @swetha-paladugu-oyo what you describe really does sound like you've failed to set the event mask. The controller might indeed be establishing a connection but it's unable to tell you because you haven't unmasked the HCI event for it.

[CreateConnection_with_EventMask_nrf.txt](https://github.com/zephyrproject-rtos/zephyr/files/3662253/CreateConnection_with_EventMask_nrf.txt)
[CreateConnection_with_EventMask_ref.txt](https://github.com/zephyrproject-rtos/zephyr/files/3662254/CreateConnection_with_EventMask_ref.txt)

Hi @jhedberg @carlescufi 
We tried the same experiment by providing LE_Set_Event_Mask with "LE Connection Complete Event | LE Advertising Report Event | LE Connection Update Complete Event | LE Read Remote Features Complete Event | LE Long Term Key Request Event"
Even after this we are not getting the events
Could you please have a look at the log in case if we are missing on some steps

jhedberg, 2019-09-27:
@swetha-paladugu-oyo you also need to enable the LE Meta event using the generic event mask HCI command. Otherwise it doesn't matter what you set to the LE Event mask.

carlescufi, 2019-09-27:
@swetha-paladugu-oyo after a quick look at your logs I see that you are not issuing `Read Buffer Size` to the controller. That is *mandatory* as per spec. I don't think it's related but you should do it anyway.

swetha-paladugu-oyo, 2019-09-27:
Hi @carlescufi , @jhedberg 
After setting LE_Meta_event mask we are able to see the connection complete events.
Thank you so much for your help. We wouldn't have solved it without your help.

carlescufi, 2019-09-27:
@swetha-paladugu-oyo glad to hear that! Can you close the issue please?

swetha-paladugu-oyo, 2019-09-27:
Sure.

closed 2019-09-27 by swetha-paladugu-oyo