zephyrproject-rtos/zephyr #19414: UART and prf not working

ErekoseTCE, 2019-09-26:
** Description **
I was porting some code from the nRF52840 to the STMF7 and ran into a crash.  
To get to the crash I had to bypass a compiler optimization issue with prf.c that was overwriting my global variables.  
I describe what I did to bypass the prf.c issue (which you may not be able to do anything about) to get to the crash (something you might).

Attached is my code for both sides of a serial connection between a Nucleo F746ZG and a Mac.  
STM Setup:
    The Mac is connected to the Nucleo 1) via a USB cable to the ST-Link microB and 2) via a USB cable to a USB-to-UART (CP2102) attached to CN9 pins 4,6,8,10 and uses HW flow control
nRF Setup:
    The Mac is connected to the nRF52840-PCA10059 1) via a USB cable to a Segger J-Link to the SWO and SCLK and 2) via a USB cable to a USB-to-UART (CP2102) attached to pins 2,29,31,10 and uses HW flow control

Attached is a diff file of the small number of changes I made to the zephyr dts files to make this work.

**To Reproduce**
Steps to reproduce the behavior:
1. unzip attached source code
2. cd STM_uart
3. west build
4. vi build/build.ninja and edit the compiler flags for prf.c.  Change -Og to -O0
5. screen /dev/tty.usbmodem123545 115200 to see log output
6. west debug
7. c

8. In another terminal, make -f runtest.make
9. ./runtest /dev/cu.SLAB_TO_UART
10. See in log output that the STM has crashed

**Expected behavior**
Not to crash but instead runtests reports success.   If you repeat these steps on the nRF52840 it works. (note that the prj.conf needs some lines commented and uncommented)

**Impact**
almost a showstopper.  We can't continue developing on the STM which can do things the nRF can't.  We are going to a plan b for the nRF but that's not prefered.

**Screenshots or console output**

**Environment (please complete the following information):**
 - OS: MacOS
 - Toolchain (e.g Zephyr SDK,ZEPHYR_TOOLCHAIN_VARIANT="gnuarmemb")
 - zephyr git commit 3c82d233e3f9cd74e72cadd60e435f835ab46607

**Additional context**

[stm_bug_diffs.txt.zip](https://github.com/zephyrproject-rtos/zephyr/files/3659216/stm_bug_diffs.txt.zip)
[nucleo-f746zg-uart-bug.tgz.zip](https://github.com/zephyrproject-rtos/zephyr/files/3659217/nucleo-f746zg-uart-bug.tgz.zip)

erwango, 2019-09-26:
Thanks for reporting this issue.
FYI, it might take some time for me to have the time to build your project and debug in real conditions. Having a zephyr sample (if doable on your side) would be much faster for you to get a feedback. Meanwhile, can you provide log out put directly here?

ErekoseTCE, 2019-09-26:
> Thanks for reporting this issue.
> FYI, it might take some time for me to have the time to build your project and debug in real conditions. Having a zephyr sample (if doable on your side) would be much faster for you to get a feedback. Meanwhile, can you provide log out put directly here?

I didn't see a interrupt driven uart sample and I think that is key to the problem.  The hello world app works fine.

The log output is very limited (open to suggestion on how to add more).  Furthermore I have noticed some differences in how it crashes, i.e. different USAGE FAULT text sometimes.   The last output was:
***** Booting Zephyr OS zephyr-v1.14.0-175-g3c82d233e3f9 *****
[Setup_uart]  initial settings: baud: 115200, parity: 0, stop: 1, data: 3, flow: 0
[Setup_uart]  initial settings: baud: 115200, parity: 0, stop: 1, data: 3, flow: 1
[main]  Main loop 1
[main]  Main loop 2
[main]  Main loop 3
[main]  Main loop 4
***** USAGE FAULT *****
  Illegal use of the EPSR
***** Hardware exception *****
Current thread ID = 0x200007a8
Faulting instruction address = 0x0
Fatal fault in essential thread! Spinning...

erwango, 2019-11-01:
@ErekoseTCE, is the issue still active?

erwango, 2020-01-20:
^^ @ErekoseTCE

erwango, 2020-09-16:
Stale for nearly one year. Closing

closed 2020-09-16 by erwango