zephyrproject-rtos/zephyr #19415: typo in nucleo_l496zg.dts

ErekoseTCE, 2019-09-26:
**Describe the bug**
When looking for another problem I noticed that this dts file refers to usart2_pins_d and I don't think this is such a thing, but there is a usart2_pins_b.   I don't have this board to verify.

**To Reproduce**
Steps to reproduce the behavior:
1. do a global search for usart2_pins_d in the zephyr src tree
2.Notice that it is only found in one place which is not correct and you will get dts compile issues

**Expected behavior**
all pin references should be present

**Impact**
annoyance for anyone trying to use that uart

**Screenshots or console output**
```
diff --git a/boards/arm/nucleo_l496zg/nucleo_l496zg.dts b/boards/arm/nucleo_l496zg/nucleo_l496zg.dts
index bc69847856..f3239c9b48 100644
--- a/boards/arm/nucleo_l496zg/nucleo_l496zg.dts
+++ b/boards/arm/nucleo_l496zg/nucleo_l496zg.dts
@@ -65,7 +65,7 @@ arduino_serial: &lpuart1 {};

 &usart2 {
        current-speed = <115200>;
-       pinctrl-0 = <&usart2_pins_d>;
+       pinctrl-0 = <&usart2_pins_b>;
        pinctrl-names = "default";
        status = "ok";
 };
```
**Environment (please complete the following information):**
 - OS: MacOS
 - Toolchain (e.g Zephyr SDK,ZEPHYR_TOOLCHAIN_VARIANT="gnuarmemb")
 - zephyr git commit 3c82d233e3f9cd74e72cadd60e435f835ab46607

**Additional context**

erwango, 2019-09-30:
@ErekoseTCE, after a quick search it seems `usart2_pin_d` is actually valid and defined:
https://github.com/zephyrproject-rtos/zephyr/blob/bfa89782a86b23494504efc36387e8487095638a/dts/arm/st/l4/stm32l4-pinctrl.dtsi#L49

Then, fyi, sadly these references are not yet used in Zephyr and you should actually rely on inmux.c file. Device tree pinmux definitions have been pushed before the code to actually use them and due to some changes that happen in between, the code never went in and they are finally not actually used. Still, some people use these out of tree and this is why they are maintained. Which is the reason for this unclear status. Sorry for that.

ErekoseTCE, 2019-09-30:
My bad, I am not sure how it happened but a search in my tree did not report that entry, but then I noticed that a few dts files were not present in my tree.  I have no idea why they were missing, but a fresh checkout restored them and now I see the references you mention.    As far as I can tell those dtsi files were the only ones missing, so I must have done but can't figure out how.   Anyway sorry for the false alarm

closed 2019-09-30 by ErekoseTCE