zephyrproject-rtos/zephyr #19419: Offer a simpler alternative to West

cerealkill, 2019-09-26:
**Is your feature request related to a problem? Please describe.**
I need to automate build for multiple platforms but West does not provide scriptability, only command line execution. Nor does the documentation make the build process clear and manually reproducible.

**Describe the solution you'd like**
Making west a build automation tool.

**Describe alternatives you've considered**
Better compilation documentation enabling the user to fully comprehend environment variables and steps.

**Additional context**

carlescufi, 2019-09-27:
@cerealkill have you looked at `scripts/sanitycheck`?

nashif, 2019-11-20:
@cerealkill  Can you please describe your usecase in more details?

nashif, 2019-11-29:
Please reopen with more details, not sure what is being requested here and probably it is out of scope and needs to be scripted based on the usecase.

closed 2019-11-29 by nashif