zephyrproject-rtos/zephyr #19420: power: system power management does not know how long to sleep for

INemesisI, 2019-09-26:
**Is your enhancement proposal related to a problem? Please describe.**
When going into low power states (non deep sleep), the system relies on the SoC specific function `sys_set_power_state(enum power_states state)` to go into sleep.
The calling function relies on the SoC function to go into sleep for the given time and will resume normal execution right after calling it.
However the SoC function does not have any information about how long to go into lower power modes for, since the time is not passed to the function. 

Currently the best assumption that can be done is that for the given sleep mode the device needs to be sleeping for at least `CONFIG_SYS_PM_MIN_RESIDENCY_SLEEP_X` ticks. This defines the minimal threshold for when to go into a specific sleep mode X. 
This will cause the device to always wake up earlier than it needs to and waste power.

**Describe the solution you'd like**
The number of ticks, the system is allowed to sleep for is passed to the calling function ` _sys_suspend(s32_t ticks)`.
The easy solution would be to pass `ticks` also to `sys_set_power_state`.
Most sleep implementations will probably utilise the internal RTC to wake up so converting the ticks to a time in milliseconds would probably make more sense so that the SoC specific implementation does not need to worry about what a system tick zephyr uses.
milliseconds = ticks * MSEC_PER_SEC / SECS_TO_TICKS;
So I would suggest to change the function to `void sys_set_power_state(enum power_states state, u32_t milliseconds)`

I would be interested in what the zephyr developers think about this solution or If there is maybe a more elegant way that I am just overseeing here.

aescolar, 2019-09-27:
CC @andyross @pizi-nordic

pizi-nordic, 2019-09-27:
I acknowledge the problem (see https://github.com/zephyrproject-rtos/zephyr/issues/14306) however in my opinion the sys_set_power_state() is not a right place to implement this feature.

The proposed solution will make the SoC code responsible for:
1. Managing timer used to wake-up from low-power mode.
2. Taking under account potential wake-up latency and scheduling a timeout earlier to mitigate it.

I think that the first item is the most problematic: In low power mode you might loose context of default system timer (like systick), which you have to restore basing on the time counted in the RTC or other peripheral which tracked time while system was suspended.

I think that the better place is to implement this feature is the system layer, especially if you take under account a fact that the RTC used for wake-up could be external device located on SPI/I2C bus and the OS should must advance system timer after wake-up.

INemesisI, 2019-09-27:
Oh yes, I see your point. 
Having a timer implementation trigger a wakeup gives us much more flexibility, especially when taking external wakeup sources into account.

Realising that my proposal really makes no sense, thank you for the clarification!

closed 2019-09-27 by INemesisI