zephyrproject-rtos/zephyr #19432: nrfx: nrf52840_pca10056 SPIM1 cannot be selected without SPIM3

paulgavrikov, 2019-09-27:
**Describe the bug**
Enabling SPIM1 for nrf52840_pca10056 seems to enable `NRFX_SPIM_EXTENDED_ENABLED` which is only available for SPIM3. If SPIM3 is not activated the project will not compile.

**To Reproduce**
Steps to reproduce the behavior:
overlay:
```
&i2c0 {
    /* Cannot be used together with spi0. */
	status = "disabled";
};

&i2c1 {
	/* Cannot be used together with spi1. */
	status = "disabled";
};

&spi0 {
	/* Cannot be used together with i2c0. */
	status = "okay";
	compatible = "nordic,nrf-spis";
	sck-pin = <28>;
	mosi-pin = <29>;
	miso-pin = <30>;
	csn-pin = <31>;
	def-char = <0x00>;
};

&spi1 {
	/* Cannot be used together with i2c1. */
	status = "okay";
	compatible = "nordic,nrf-spim";
	sck-pin = <47>;
	mosi-pin = <3>;
	miso-pin = <2>;
	/* csn-pin = <46>; DT compiler does not parse master CSN*/
};
```

prj.conf:
```
CONFIG_SPI=y
CONFIG_SPI_NRFX=y
CONFIG_SPI_ASYNC=y
CONFIG_SPI_SLAVE=y
CONFIG_SPI_0=y
CONFIG_SPI_0_NRF_SPIS=y
CONFIG_SPI_1=y
CONFIG_SPI_1_NRF_SPI=y
CONFIG_SPI_1_NRF_SPIM=y
CONFIG_SPI_2=n
CONFIG_SPI_3=n
```

**Expected behavior**
SPIM1 should be selectable independent of SPIM3. 

**Impact**
Showstopper.

**Screenshots or console output**
```
/home/ntrnd/subversion/components/zephyr/system/Zephyr/modules/hal/nordic/nrfx/drivers/src/nrfx_spim.c:42:2: error: #error "Extended options are available only in SPIM3 on the nRF52840 SoC."
 #error "Extended options are available only in SPIM3 on the nRF52840 SoC."
  ^~~~~
/home/ntrnd/subversion/components/zephyr/system/Zephyr/modules/hal/nordic/nrfx/drivers/src/nrfx_spim.c: In function 'nrfx_spim_init':
/home/ntrnd/subversion/components/zephyr/system/Zephyr/modules/hal/nordic/nrfx/drivers/src/nrfx_spim.c:180:38: error: 'NRFX_SPIM3_INST_IDX' undeclared (first use in this function); did you mean 'NRFX_SPIM1_INST_IDX'?
     if ((p_instance->drv_inst_idx != NRFX_SPIM3_INST_IDX) &&
                                      ^~~~~~~~~~~~~~~~~~~
                                      NRFX_SPIM1_INST_IDX
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows): Kubuntu 18.04
 - Toolchain (e.g Zephyr SDK, ...): Zephyr SDK 0.10.3
 - Commit SHA or Version used: 45dba1e1f04d3968bdf19a38951f1c890e287dd6

**Additional context**
Add any other context about the problem here.

anangl, 2019-09-27:
Fixed by #19399.

paulgavrikov, 2019-09-27:
Works now. Thanks.

closed 2019-09-27 by paulgavrikov