zephyrproject-rtos/zephyr #19435: how to change uart tx rx pins in zephyr

swetha-paladugu-oyo, 2019-09-27:
Hi,

I'm using nordic nrf52810 development board and flashed hci_uart code of zephyr to communicate over HCI commands.
Default zephyr uses pins 6 and 8 of uart0 instance. But unfortunately the nordic has other uart tx and rx pins.
Can you please let me know where to change the tx and rx pins in zephyr code?

carlescufi, 2019-09-27:
> Hi,
>
> I'm using nordic nrf52810 development board and flashed hci_uart code of zephyr to communicate over HCI commands.
> Default zephyr uses pins 6 and 8 of uart0 instance. But unfortunately the [...]

You might want to consider the [mailing list](https://lists.zephyrproject.org/) or [Slack](https://tinyurl.com/y5glwylp) for these kind of questions.

You need to use [Device Tree](https://docs.zephyrproject.org/latest/guides/dts/index.html)

swetha-paladugu-oyo, 2019-09-30:
Hi @carlescufi 
I have changed the uart pins in file nrf52810_pca10040.dts in zephyrproject/zephyr/boards/arm/nrf52810_pca10040 folder.
And it worked..
But want to know if it is the correct place and only place where the pins has to be changed??

carlescufi, 2019-09-30:
> Hi @carlescufi
> I have changed the uart pins in file nrf52810_pca10040.dts in zephyrproject/zephyr/boards/arm/nrf52810_pca10040 folder.
> And it worked..
> But want to know if it is the correct place and only place where the pins has to be changed??

No, you can also create your own board, and with it your own set of `.dts` files, or you can also use DT overlays, please see the documentation or look for files named `.overlay`.

closed 2020-10-06 by github-actions[bot]