zephyrproject-rtos/zephyr #19437: tests/kernel/sched/schedule_api tests fail to build

galak, 2019-09-27:
When we do:
`./scripts/sanitycheck -p qemu_cortex_m3 -T tests/kernel/sched/schedule_api/`

We get:

```
4 tests selected, 804 tests discarded due to filters
total complete:    0/   4   0%  failed:    0

qemu_cortex_m3            tests/kernel/sched/schedule_api/kernel.sched       FAILED: build_error
	see: sanity-out/qemu_cortex_m3/tests/kernel/sched/schedule_api/kernel.sched/build.log

total complete:    1/   4  25%  failed:    1

qemu_cortex_m3            tests/kernel/sched/schedule_api/kernel.sched_no_timeslicing FAILED: build_error
	see: sanity-out/qemu_cortex_m3/tests/kernel/sched/schedule_api/kernel.sched_no_timeslicing/build.log
```

```
/home/galak/git/zephyr/kernel/sched.c:759:3: error: #error Too many priorities for multiqueue scheduler (max 32)
 # error Too many priorities for multiqueue scheduler (max 32)
   ^~~~~
```

Not sure if we need to have 'CONFIG_SCHED_SCALABLE=y' enabled in `tests/kernel/sched/schedule_api/prj.conf` or something else.

andyross, 2019-10-02:
Yes.  The multiqueue scheduler has a hard limit of 32 priorities, it's the default for qemu_cortex_m3, and this test by default uses more (for no particularly good reason, IIRC).  There is a prj_multiq.conf kconfig file that can be used to exercise that scheduler backend, or you can use a different one.

At one point, that was a default platform and passed this test, presumably by an ancestor of the current trick.  It's gotten bitrotten since being disabled I guess.

andyross, 2019-10-02:
The most minimal fix is submitted above.  Other alternatives:

1. Change the default test config to not use so many priorities needlessly.

2. Use a blacklist for qemu_cortex_m3 instead of the filter (filters have performance implications)

3. Change the default kconfig for qemu_cortex_m4 to remove SCHED_MULTIQ.  That was done for test coverage originally, but as it's not a default platform any more there's no good reason to have it there.

PR #19565 (2019-10-03): tests/kernel/sched/schedule_api: Filter for MULTIQ platforms
This test requires more than 32 static priorities by default, and
doesn't run with the multiq scheduler without a special configuration.
That used to be specified per-platform, but got moved to a separate
test case a while back.

This broke non-default platforms like qemu_cortex_m3 which use
SCHED_MULTIQ as their default backend.  Put a filter in place instead
of going back to per-platform changes.

Fixes #19437
commit bbd910b2 (2019-10-03): tests/kernel/sched/schedule_api: Filter for MULTIQ platforms

This test requires more than 32 static priorities by default, and
doesn't run with the multiq scheduler without a special configuration.
That used to be specified per-platform, but got moved to a separate
test case a while back.

This broke non-default platforms like qemu_cortex_m3 which use
SCHED_MULTIQ as their default backend.  Put a filter in place instead
of going back to per-platform changes.

Fixes #19437
closed 2019-10-03 by nashif