zephyrproject-rtos/zephyr #19447: SEGGER_RTT.h: No such file or directory

rajputh54, 2019-09-28:
I'm trying to compile an example 
                       samples/subsys/logging/logger/
by adding below features in prj.conf
CONFIG_LOG=y
CONFIG_LOG_RUNTIME_FILTERING=y
CONFIG_LOG_BUFFER_SIZE=2048
CONFIG_LOG_PRINTK=y
CONFIG_LOG_PROCESS_TRIGGER_THRESHOLD=0
CONFIG_LOG_BACKEND_RTT=y
CONFIG_USE_SEGGER_RTT=y
CONFIG_COVERAGE=n

but getting error 
[ 33%] Building C object zephyr/CMakeFiles/zephyr.dir/subsys/logging/log_backend_rtt.c.obj
/home/hardik/practice/nrf52/fw-nrfconnect-zephyr/subsys/logging/log_backend_rtt.c:11:10: fatal error: SEGGER_RTT.h: No such file or directory
 #include <SEGGER_RTT.h>
          ^~~~~~~~~~~~~~
compilation terminated.

I tried to find SEGGER_RTT.h in all over the code but there is no such file.

ydamigos, 2019-09-28:
Hi,

`SEGGER_RTT.h` is located under `modules/debug/segger/rtt/SEGGER_RTT.h`
Did you use [west tool](https://docs.zephyrproject.org/latest/guides/west/index.html) to get the modules?

rajputh54, 2019-09-30:
I got the `modules/debug/segger/rtt/` directory using the west tool.

Thanks

closed 2019-09-30 by rajputh54