zephyrproject-rtos/zephyr #19459: Bluetooth: Mesh:  Mesh Model State Binding.

menglingao1997, 2019-09-29:
In **zephyr/samples/boards/nrf52/mesh/onoff_level_lighting_vnd_app/**

According Mesh Model Spec.
Generic Power OnOff Server has Generic Power Actual state **bindings** with the Generic OnoOff State.
Light Lightness Server has Light Lightness Actual State also **bindings** with the Generic OnOff State.

https://github.com/zephyrproject-rtos/zephyr/blob/c0c4ba85164e7ba2351315e763f0c3b9a0473ac9/samples/boards/nrf52/mesh/onoff_level_lighting_vnd_app/src/mesh/state_binding.c#L284-L288.

As shown in the code above,
When the state of lightness actual state changes, we will also change Generic OnOff States. But I did not find any changes to the Generic Power OnOff State.

closed 2019-09-29 by menglingao1997

jhedberg, 2019-09-29:
@vikrant8051 any comments on this?

menglingao1997, 2019-09-29:
I'm mistake. But I still think that the coupling of this example is too big to be tailored. I think it would be better if you could make state bindings more abstract.

vikrant8051, 2019-09-29:
>> Generic Power OnOff Server has Generic Power Actual state bindings with the Generic OnOff
>> State
>>Light Lightness Server has Light Lightness Actual State also bindings with the Generic OnOff
>> State.

1. Generic Power OnOff Server has **Generic OnPowerUp** state.

2. A Generic Power OnOff Server shall use the Generic OnPowerUp state to determine the behavior after a node is **powered up**.

3. Value of Generic ONPowerUp state should not get vary as we vary Gen. OnOff stae or Light Lightness Actual State. Because only user can change it (for e.g via SmartPhone Mesh App which supports  Generic Power OnOff Client) or manufacturer fixed it any value out of 0x00, 0x01 or 0x02.
        
4. Genric OnOff & Light Lightness Actual State are bounded bidirectinally.

   But same is not true for Generic OnPowerUp state with any other state.

   It is only one directional binding which execute only when device Boot up or get power.

   For details, please refer Table 3.6 of Mesh Model Specification.

   As per it, after device gets power on, algorithm reads value of Generic OnPowerUp state from persistence storage, & adjust all states associated with Lightness (Genric OnOff -> Generic Level -> Light Lightness Actual -> Light Lightness Linear -> Light CTL Lightness).

So it is not a bug.

vikrant8051, 2019-09-29:
> I'm mistake. But I still think that the coupling of this example is too big to be tailored. I think it would be better if you could make state bindings more abstract.

@menglingao1997 
It is already represented in most simplified way compared to what Bluetooth Mesh Model Specification has stated. Architecture has designed with lots of care & thorough testing (including PTS tests). Several times I had reorganized the things to make it more simple, robust & bug free implementation. There was/is no reference openly available since all available stacks from various Semiconductor Vendors are closed source/black box. 

Kindly understand full implementation first & help me to optimized it further. But condition are, it should not add unnecessary burden on algorithm & bug free. 

Because personally I am not happy with Bluetooth Mesh's Group of Nodes performances (This is true even with any available Market Solutions based on Bluetooth Mesh ).

Regards,
Vikrant

vikrant8051, 2019-10-29:
Hi @menglingao1997,
https://github.com/vikrant8051/zephyr/tree/fix_bugs32/samples/boards/nrf52/mesh/onoff_level_lighting_vnd_app

Please go through with above project which is improved version of onoff_level_lighting_vnd_app.

(https://github.com/zephyrproject-rtos/zephyr/pull/20170 this is PR which will merged soon to reflect the changes.)

Hope you will like it. Any feedback or suggestions are highly welcome.

Thank You !!