zephyrproject-rtos/zephyr #19467: Error building samples with PWM

mron, 2019-09-30:
**Describe the bug**
A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context about the problem here.

mron, 2019-09-30:
Tell me what PWMS_CONTROLLER is and where to look for it, I'll see if I can find the problem.

ValentiWorkLearning, 2019-10-02:
@mrrosen  see answer here: https://github.com/zephyrproject-rtos/zephyr/issues/19523

mrrosen, 2019-10-02:
@ValentiWorkLearning I think you have the wrong "@"; @mron see above for help

closed 2020-10-05 by github-actions[bot]