zephyrproject-rtos/zephyr #19472: usb_dc_stm32: usb shows after some time errors and warnings #stm32 #usb #usb_dc

StefJar, 2019-09-30:
Looks like that I can manage to generate a warning/error on the usb device.
I am doing rx and tx async. I put a code snippet below which shows the important part of my code.

Setup:
```
CONFIG_USB=y
CONFIG_USB_DC_STM32=y
CONFIG_USB_DEVICE_STACK=y
CONFIG_USB_DEVICE_VID=0x2FE4
CONFIG_USB_DEVICE_PID=0x1
CONFIG_USB_DEVICE_MANUFACTURER="kt"
CONFIG_USB_DEVICE_PRODUCT="dev1"
CONFIG_USB_CDC_ACM=y
CONFIG_USB_VBUS_GPIO=y
CONFIG_USB_VBUS_GPIO_DEV_NAME="GPIOA"
CONFIG_USB_VBUS_GPIO_PIN_NUM=9
```

Shown error:
```
[00:00:32.941,000] <err> usb_dc_stm32: Unable to get write lock (-16)
[00:00:32.948,000] <wrn> usb_device: Failed to write endpoint buffer 0x82
[00:00:32.956,000] <err> usb_dc_stm32: Unable to get write lock (-16)
[00:00:32.963,000] <wrn> usb_device: Failed to write endpoint buffer 0x82
```

Code snippet:

```
typedef struct usbACMs {
	struct device * uart;

	// rtfw-I
	struct {
		rtfw_proto_rx_t rx;
		rtfw_proto_io_t proto;
		struct ring_buf rxRB;
		struct k_sem rxSem;
		u8_t rxBuffer [USBACM_RX_BYTES_AMOUNT];
	} rtfwI;

	struct {
		u8_t buffer[32];
	} rx;

	struct {
		unsigned int N;
		u8_t * pD;
		struct k_sem done;
		struct k_mutex guardM;
	} tx;
} usbACM_t;

static usbACM_t usbACM;

static void usbACM_isr(struct device *dev) {
	int am;

	uart_irq_update(dev);

	if (uart_irq_tx_ready(dev)) {
		k_sem_give(&usbACM.tx.done);
	}

	if (uart_irq_rx_ready(dev)) {
		while(true) {
			am = uart_fifo_read(dev,usbACM.rx.buffer, sizeof(usbACM.rx.buffer));
			if (am) {
				if (am != ring_buf_put(&usbACM.rtfwI.rxRB, usbACM.rx.buffer, am)) {
					LOG_ERR("usb acm rx buffer overflow");
				}
			} else {
				break;
			}
		}
		k_sem_give(&usbACM.rtfwI.rxSem);
	}
}

static void usbACM_tx_helper_sendData(u8_t * pD, u8_t N) {
	int am;

	k_mutex_lock(&usbACM.tx.guardM, K_FOREVER);
	uart_irq_tx_enable(usbACM.uart);

	while (N) {
		am = uart_fifo_fill(usbACM.uart, (const u8_t *)pD, N);
		k_sem_take(&usbACM.tx.done, K_FOREVER);
		N -= (u8_t)am;
		pD += am;
	}
	uart_irq_tx_disable(usbACM.uart);
	k_mutex_unlock(&usbACM.tx.guardM);
}

```

jfischer-phytec-iot, 2019-09-30:
Can you reproduce the issue with `samles/subsys/usb/cdc_acm`? If true, is is related to #19038 and #16673?

StefJar, 2019-09-30:
I am quite busy and I have no dev board here. My code is old and I created it based on the example.

I had a similar issue in Feb 2019. Different code but similar problem. 

https://github.com/zephyrproject-rtos/zephyr/issues/13799

Maybe something changed and someone else have a similar problem

jfischer-phytec-iot, 2019-09-30:
> I am quite busy and I have no dev board here. My code is old and I created it based on the example.
> 
> I had a similar issue in Feb 2019. Different code but similar problem.
> 
> #13799
> 
> Maybe something changed and someone else have a similar problem

okay, I close it then. Please reopen when you can reproduce the issue with `samles/subsys/usb/cdc_acm` on latest master and it is not related to #19038 and #16673?

closed 2019-09-30 by jfischer-phytec-iot