zephyrproject-rtos/zephyr #19473: Missing NULL parameter check in k_pipe_get

koffes, 2019-09-30:
**Describe the bug**
MPU fault occurs if the data parameter to k_pipe_get is NULL.

**Expected behavior**
Would expect NULL to be a way to flush the pipe. E.g., we don't want the actual data, throw it away. Optionally, create a separate flush API and return an error if data is NULL.

andyross, 2019-09-30:
This seems more like a feature request for a flush API, not a missing parameter check.  We don't document exactly what happens if the buffer isn't valid, but crashing is IMHO not a terrible choice.  It works for memcpy() after all.

I don't disagree that some kind of flush behavior should be supported, but that's an enhancement request, not a bug.

koffes, 2019-10-02:
@andyross : It is a high-level design decision, but I would prefer a null pointer check and a proper error message or error code printout in all modules before getting a MPU_FAULT.  I have no problems finding the cause of the MPU_FAULT, but I don't think we should expect all users to see the faulting instruction address, do an `addr2line`, then set a breakpoint on that line and run the program again. This is time consuming and difficult for many users. 

This discussion is beyond the scope of this issue. If you want, I can close this and open 1: A general issue on parameter checks and 2: an API request for flushing.

d3zd3z, 2019-12-13:
At one point, not having argument checks was documented specifically as a design philosophy of Zephyr. I can't find this anywhere in the current documentation (other than a mention of this in the syscalls documentation), but I think it is still a general principle use to keep Zephyr usable on a small footprint.

Maybe the argument check could be an assertion, which would then get compiled out of production builds?

koffes, 2019-12-16:
@d3zd3z I think that is a good idea.

andrewboie, 2019-12-16:
The ask here if I understand correctly, is really that all Zephyr APIs that take a pointer value should have a NULL check for those pointer values. k_pipe shouldn't be a special case.

Personally I don't see a lot to be gained by this, it will add extra footprint/overhead for these checks, and there are of course many other bogus pointer values that would not be caught by this.

> We don't document exactly what happens if the buffer isn't valid, but crashing is IMHO not a terrible choice. It works for memcpy() after all.

Pretty much my feeling here.

Having said all that, if we want to add a flushing mechanism to k_pipe, no objections at all, seems reasonable and useful.

d3zd3z, 2019-12-17:
I guess I wasn't really trying to argue for adding the null check (I don't think we should have it here), just that if we did add one, it shouldn't be in production builds.

It would add a lot of code to Zephyr if we checked arguments anywhere.

We will have to decide how to deal with the MISRA C requirement to have them, though.

andrewboie, 2019-12-17:
> I guess I wasn't really trying to argue for adding the null check (I don't think we should have it here), just that if we did add one, it shouldn't be in production builds.
>
> It would add a lot of [...]

Yeah, just looked this up...Directive 4.1 "Unless a pointer is known to be non-NULL, a run-time check should be made before dereferencing that pointer". I wonder if this is a 'should' vs 'must' kind of situation. @ceolin might know

d3zd3z, 2019-12-17:
There are two ways we could deal with Dir 4.1. "Unless a pointer is known to be non-NULL" could be argued that is something Coverity checks. Or the directive is "required", which is MISRA C speak for "should". It is required, unless we write down why we don't comply with it. The wording says it is hard to reason across function boundaries, but something like Coverity is good at that.

koffes, 2019-12-18:
I have noticed that the Zephyr community has some reservations towards macros, but what about this:
NULL_PARAM_ASSERT(arg1, arg2, ...)
Such a "vararg" macro can be enabled by default in debug mode and lead to an assertion.
In case of many potential NULL arguments this would avoid having an array of assert checks at the top of a function.

andrewboie, 2019-12-18:
I think this ticket is overloaded, we have two issues

1) a request to add a flushing mechanism to k_pipe API
2) considerations on globally adding NULL checks to zephyr APIs

I think this should be split into two tickets

koffes, 2019-12-19:
k_pipe flush: #21512 
NULL checks issue  #21513
Closing this issue.

closed 2019-12-19 by koffes