zephyrproject-rtos/zephyr #19484: Bluetooth:  bt_set_name() asserts due to flash and radio coex

paulgavrikov, 2019-09-30:
**Describe the bug**
Calling `bt_set_name()` **sometimes** fails even though no radio activity is present.
This appears to be only happening when a LE scan was started before (and stopped). Independent of the duty cycle.

**To Reproduce**
[image]

Happens after a couple of loops on nrf52840_dks. Only happens with `test1()` and `test2()`. Removing either of them appears to never run in the assertion.

**Expected behavior**
No assert.

**Impact**
showstopper.

**Screenshots or console output**
```
ASSERTION FAIL [ll_radio_state_is_idle()] @ ZEPHYR_BASE/drivers/flash/soc_flash_nrf.c:291␍␊
[15:04:10:857] ⇥    Radio is on during flash operation.␍␊
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows): Kubuntu 18.04
 - Toolchain (e.g Zephyr SDK, ...): Zephyr SDK 0.10.3
 - Commit SHA or Version used: 2051ad184614f02975ea8ba923cf5bff13c2535a

**Additional context**
Add any other context about the problem here.

paulgavrikov, 2019-09-30:
@cvinayak

carlescufi, 2019-09-30:
@paulgavrikov please let us know if https://github.com/zephyrproject-rtos/zephyr/pull/19395 fixes the issue.

paulgavrikov, 2019-09-30:
@carlescufi #19395 does not fix this

paulgavrikov, 2019-09-30:
Actually I am not sure if `test2()` waits for the discovery callback to be triggered. Could it be possible that a stop call is not processed if the start was issued shortly before?

cvinayak, 2019-10-01:
@paulgavrikov I tried the same scenario manually using the bluetooth shell app, but could not reproduce. I tried to modify the scan_adv sample to perform the scenario by enabling settings and flash features in the sample, but could not manage to get the app to flash (and hence not reproduce your scenario). 

Option 1: provide a very simple application that reproduces the issue.
Option 2: we try a screen sharing call to debug the issue live (between 0600 to 1400 CET)

cvinayak, 2019-10-04:
@paulgavrikov I may have managed to reproduce something similar manually (though with difficulty, like 1 in 20 times) trying to use ```flash erase 60000``` while having ```bt scan on``` with bt shell app and 2.5 ms continuous scan.

I have updated https://github.com/zephyrproject-rtos/zephyr/pull/19395

paulgavrikov, 2019-10-04:
@cvinayak Appears to be working with #19395! Thank you very much!

PR #19395 (2019-10-04): Bluetooth: controller: split: Fix flash driver co-operation
When flash driver requests abort of radio event in unreserved
time space, resume radio events in the pipeline where not
flushed. These resumed events caused flash driver to assert
on the check whether radio was in use.

Fixed by flushing the pipeline of all radio events, resume
and also those events in pipeline with pre-empt timeout
being setup.

Fixes: #19484.
closed 2019-10-04 by paulgavrikov