zephyrproject-rtos/zephyr #19490: Bluetooth: split: 'e' assert during disconnect

paulgavrikov, 2019-09-30:
**Describe the bug**
When using the Split LL with Zephyr we occasionally observe crashes during different operations on master or slave side. Upon closer observation it looks like these asserts happen after a disconnect packet was sent. Reducing the maximum number of supported connecations does not fix this. Switching back to the legacy LL resolves this.

**To Reproduce**
Our ZTests connect to a peripheral and typically perform a couple of actions (e.g. GATT ops, pairing, L2CAP etc.). Then we disconnect and run the next test. I haven't found one specific test that always fails as the asserts appear to be happen randomly. 

**Expected behavior**
The stack should not crash.

**Impact**
showstopper

**Screenshots or console output**
```
[20:56:06:560] assert: 'e' failed␍␊
[20:56:06:571] [00:01:49.082,061] <0x1b>[1;31m<err> os: r0/a1: 0x00000003 r1/a2: 0x0000000a r2/a3: 0x00000001<0x1b>[0m␍␊
[20:56:06:571] [00:01:49.090,759] <0x1b>[1;31m<err> os: r3/a4: 0x0002e963 r12/ip: 0x00000008 r14/lr: 0x000197ef<0x1b>[0m␍␊
[20:56:06:579] [00:01:49.099,456] <0x1b>[1;31m<err> os: xpsr: 0x61000011<0x1b>[0m␍␊
[20:56:06:594] [00:01:49.104,705] <0x1b>[1;31m<err> os: Faulting instruction address (r15/pc): 0x000197fa<0x1b>[0m␍␊
[20:56:06:594] [00:01:49.112,609] <0x1b>[1;31m<err> os: >>> ZEPHYR FATAL ERROR 3: Kernel oops<0x1b>[0m␍␊
[20:56:06:600] [00:01:49.119,476] <0x1b>[1;31m<err> os: Current thread: 0x20001200 (BT ECC)<0x1b>[0m␍␊
[20:56:06:609] [00:01:49.126,190] <0x1b>[1;31m<err> os: Halting system<0x1b>[0m␍␊
```
[image]

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows): Kubuntu 18.04
 - Toolchain (e.g Zephyr SDK, ...): Zephyr SDK 0.10.3
 - Commit SHA or Version used: 2051ad1

**Additional context**

cvinayak, 2019-10-04:
@paulgavrikov please verify using #19469.

paulgavrikov, 2019-10-04:
@cvinayak yes, that works. I see a bunch of "D: " logs now but not asserts. Thanks!

PR #19469 (2019-10-04): Bluetooth: controller: split: Fix for Tx Buffer Overflow error
When more than one simultaneous connections are active,
transmitting data packets to peer, a termination causes
host to use the flushed pending number of completed packets
count for other active connections. This is on reception
of HCI disconnection complete event. But the controller has
not yet released any of the pending enqueued Tx buffers
which was happening after the disconnection event was
dispatched to HCI layer.

The fix here is to dispatch the disconnection complete event
from the LLL context after pending Tx buffers have  [... cut]
closed 2019-10-04 by paulgavrikov