zephyrproject-rtos/zephyr #19492: sanitycheck: unreliable/inconsistent catch of ASSERTION FAILED

marc-hb, 2019-09-30:
**Describe the bug**
Since large refactoring #17544, sanitycheck does not always catch ASSERTION FAILED in test output, some tests time out instead. When it does catch it, it doesn't always report it the same way.

**To Reproduce**
Add this ASSERTION failure:
```diff
--- a/subsys/logging/log_core.c
+++ b/subsys/logging/log_core.c
@@ -1159,6 +1159,7 @@ struct k_thread logging_thread;
 static int enable_logger(struct device *arg)
 {
        ARG_UNUSED(arg);
+       __ASSERT_NO_MSG(0);
 
        if (IS_ENABLED(CONFIG_LOG_PROCESS_THREAD)) {
                /* start logging thread */
```

Run this:
```
cd tmp/
time $ZEPHYR_BASE/scripts/sanitycheck -T $ZEPHYR_BASE/tests/kernel/common/ -v -p native_posix -p native_posix_64 -p mps2_an385 -p mps2_an521 -p qemu_x86 -p qemu_x86_64 -p qemu_x86_coverage -p qemu_cortex_m3 -p qemu_x86_nommu -p qemu_nios2 -p qemu_x86_long -p qemu_xtensa
```

**Expected behavior**
All failures are reported timely and consistently.

**Impact**
Longer test duration in some cases.
Missed failures in the worst cases: some tests have been seen printing an ASSERT and then PASS.

**Screenshots or console output**
```
2 tests selected, 382 tests discarded due to filters
 1/22 native_posix              tests/kernel/common/kernel.common.misra            FAILED: error (native 0.006s)
	see: sanity-out/native_posix/tests/kernel/common/kernel.common.misra/handler.log
 2/22 native_posix_64           tests/kernel/common/kernel.common.misra            FAILED: error (native 0.006s)
	see: sanity-out/native_posix_64/tests/kernel/common/kernel.common.misra/handler.log
 3/22 native_posix_64           tests/kernel/common/kernel.common                  FAILED: error (native 0.004s)
	see: sanity-out/native_posix_64/tests/kernel/common/kernel.common/handler.log
 4/22 native_posix              tests/kernel/common/kernel.common                  FAILED: error (native 0.008s)
	see: sanity-out/native_posix/tests/kernel/common/kernel.common/handler.log
 5/22 qemu_x86                  tests/kernel/common/kernel.common                  FAILED: handler_crash (qemu)
	see: sanity-out/qemu_x86/tests/kernel/common/kernel.common/handler.log
 6/22 qemu_x86                  tests/kernel/common/kernel.common.misra            FAILED: handler_crash (qemu)
	see: sanity-out/qemu_x86/tests/kernel/common/kernel.common.misra/handler.log
 7/22 qemu_cortex_m3            tests/kernel/common/kernel.common                  FAILED: timeout (qemu 60.059s)
	see: sanity-out/qemu_cortex_m3/tests/kernel/common/kernel.common/handler.log
 8/22 qemu_nios2                tests/kernel/common/kernel.common.misra            FAILED: timeout (qemu 60.059s)
	see: sanity-out/qemu_nios2/tests/kernel/common/kernel.common.misra/handler.log
 9/22 qemu_nios2                tests/kernel/common/kernel.common                  FAILED: timeout (qemu 60.059s)
	see: sanity-out/qemu_nios2/tests/kernel/common/kernel.common/handler.log
10/22 mps2_an385                tests/kernel/common/kernel.common                  FAILED: timeout (qemu 60.059s)
	see: sanity-out/mps2_an385/tests/kernel/common/kernel.common/handler.log
11/22 qemu_cortex_m3            tests/kernel/common/kernel.common.misra            FAILED: timeout (qemu 60.059s)
	see: sanity-out/qemu_cortex_m3/tests/kernel/common/kernel.common.misra/handler.log
12/22 mps2_an521                tests/kernel/common/kernel.common                  FAILED: timeout (qemu 60.059s)
	see: sanity-out/mps2_an521/tests/kernel/common/kernel.common/handler.log
13/22 qemu_x86_coverage         tests/kernel/common/kernel.common                  FAILED: handler_crash (qemu)
	see: sanity-out/qemu_x86_coverage/tests/kernel/common/kernel.common/handler.log
14/22 qemu_x86_64               tests/kernel/common/kernel.common                  FAILED: timeout (qemu 60.058s)
	see: sanity-out/qemu_x86_64/tests/kernel/common/kernel.common/handler.log
15/22 qemu_x86_nommu            tests/kernel/common/kernel.common                  FAILED: handler_crash (qemu)
	see: sanity-out/qemu_x86_nommu/tests/kernel/common/kernel.common/handler.log
16/22 qemu_x86_nommu            tests/kernel/common/kernel.common.misra            FAILED: handler_crash (qemu)
	see: sanity-out/qemu_x86_nommu/tests/kernel/common/kernel.common.misra/handler.log
17/22 qemu_x86_coverage         tests/kernel/common/kernel.common.misra            FAILED: handler_crash (qemu)
	see: sanity-out/qemu_x86_coverage/tests/kernel/common/kernel.common.misra/handler.log
18/22 qemu_x86_64               tests/kernel/common/kernel.common.misra            FAILED: timeout (qemu 60.059s)
	see: sanity-out/qemu_x86_64/tests/kernel/common/kernel.common.misra/handler.log
19/22 qemu_x86_long             tests/kernel/common/kernel.common                  FAILED: timeout (qemu 60.058s)
	see: sanity-out/qemu_x86_long/tests/kernel/common/kernel.common/handler.log
20/22 qemu_x86_long             tests/kernel/common/kernel.common.misra            FAILED: timeout (qemu 60.059s)
	see: sanity-out/qemu_x86_long/tests/kernel/common/kernel.common.misra/handler.log
21/22 qemu_xtensa               tests/kernel/common/kernel.common                  FAILED: timeout (qemu 60.060s)
	see: sanity-out/qemu_xtensa/tests/kernel/common/kernel.common/handler.log
22/22 qemu_xtensa               tests/kernel/common/kernel.common.misra            FAILED: timeout (qemu 60.060s)
	see: sanity-out/qemu_xtensa/tests/kernel/common/kernel.common.misra/handler.log
0 of 22 tests passed with 0 warnings in 156 seconds
```

**Environment (please complete the following information):**

Fedora 29
Zephyr SDK 0.10.3
Zephyr HEAD= fc4cb492b5106c085

**Additional context**

Long story in PR #17593

marc-hb, 2019-09-30:
Setting priority medium because of the risk of missing test failures, can hopefully be lowered once there is a root cause.

Note there's a big sanitycheck overhaul in progress #18603

nashif, 2019-11-07:
so we are concerned about tests timing out instead of just dying? 
if something asserts outside of the test, it will die and just timeout, not sure how it would be missed.

marc-hb, 2019-11-07:
> so we are concerned about tests timing out instead of just dying?

I don't see how every missed assert gracefully downgrades to a timeout, how do you know that? It's been a long time with a different codebase but I'm pretty sure I remember asserts being printed yet totally missed and the test misreported as a PASS. It depends on the how the test is implemented.

marc-hb, 2019-11-07:
>  I'm pretty sure I remember asserts being printed yet totally missed and the test misreported as a PASS

Link to the example was in the description: #17593

nashif, 2019-11-08:
> I don't see how every missed assert gracefully downgrades to a timeout, how do you know that?

an assert just put the system in panic, so test output and success is never reported, it times out.

> I'm pretty sure I remember asserts being printed yet totally missed and the test misreported as a PASS. 

you need to show me how to re-produce this reliably.

marc-hb, 2019-11-08:
Again this was a long time ago, but more precisely in https://github.com/zephyrproject-rtos/zephyr/pull/14777#issuecomment-506598640 almost EVERY qemu_x86_64 test was "reliably"/systematically ASSERTing and PASSING! What proves this not possible any more? Not a sanitycheck question. Was ASSERT not already putting the system in panic at the time?

Granted: fixing sanitycheck to stop missing ASSERTs wouldn't prove that either but it would raise the confidence level considerably.

nashif, 2019-11-12:
I do need something I can work with and something that is reproducible. Right now this looks like waving hands.

marc-hb, 2019-11-12:
What reproduction step is missing from the description?

nashif, 2020-01-16:
no updates or sings of this being a bug yet, closing.

closed 2020-01-16 by nashif