zephyrproject-rtos/zephyr #19497: log subsystem APIs need to be clearly namespaced as public or private

andrewboie, 2019-10-01:
The expectation is that the naming of functions, macros, and globals for logging follow the Zephyr coding standard that private APIs are prefixed with "z_".

Hence I would expect that symbols or macros starting with "log_" are public, and private stuff would be prefixed with "z_log_". However this appears to be inconsistently done. There are a lot of things which are scoped public but appear to be private, some examples would be log_N(), log_printf_arg_checker(), log_generic_from_user(), many macros, the list goes on.

There are also some private macros which are named with `__` leading underscores in violation of our coding standard, these should start with `Z_`.

Because of this, it's not clear what the public surface of logging is, it's clearly not everything that begins with log_, and we need to re-namespace the internal stuff to clearly indicate it is private. Among other things this will make maintenance easier since we will not have to treat the private stuff as contracts that have to go through the deprecation process if we need to change them.

Overall, I think we need something like:
- Public, application facing APIs and macros all prefixed with log_ or LOG_
- Public APIs which are used by log backend implementations all prefixed with log_backend_ or LOG_BACKEND_ (or whatever we like, just be consistent)
- Remaining internal APIs and macros all prefixed with z_log_/Z_LOG_

andrewboie, 2019-10-01:
Marking this as a bug since we have clear MISRA-C violations and we are treating APIs which are private as public.

aescolar, 2019-10-01:
CC  @nordic-krch @jakub-uC

closed 2020-12-19 by github-actions[bot]