zephyrproject-rtos/zephyr #19509: bt: stm32wb55: broken secure conenctions

pavlohamov, 2019-10-01:
Secure connection could not be established. Fails with code 1 (BT_SECURITY_ERR_AUTH_FAIL).

The first suspicious thing to look on: if BT_SMP=y then HCI will send the peer address as ff:ff:ff:ff:ff:ff (random). For the nrf52 we'll see here some valid address.
Dumped hci packet in stm32 hold the ff ff ff .. address. 

**To Reproduce**
Steps to reproduce the behavior:
1. build the peripheral_hids sample
2. cmake -GNinja -DBOARD=nucleo_wb55rg && ninja flash
3. Open serial console 
4. Try connect.
5. Enter passkey.
6. Observe authentication failure.

**Expected behavior**
Pairing successful. 

**Impact**
No secure connection could be established

**Screenshots or console output**
Connected ff:ff:ff:ff:ff:ff (random)
Security failed: ff:ff:ff:ff:ff:ff (random) level 1 err 1
Disconnected from ff:ff:ff:ff:ff:ff (random) (reason 0x13)

**Environment (please complete the following information):**
 - Linux (Ubuntu 18.04 x64)
 - Zephyr SDK
 - 5b57737817 (latest master)

**Additional context**
1) The M0 firmware is updated to latest  BLE: v1.2.0 FUS: v1.0.2
2) tried to take HAL from latest CUBE. - the same result https://github.com/zephyrproject-rtos/hal_stm32/pull/21
3) if BT_SMP is enabled. then the propr peer addresses are seen. https://github.com/zephyrproject-rtos/zephyr/pull/19508 coulde be used to simply enable/disable MP

pavlohamov, 2019-10-01:
@erwango could you be so kind to take a look?

erwango, 2019-10-01:
@pavlohamov, is that a regression ?
If yes, can you point a SHA1/setup that used to work ?

pavlohamov, 2019-10-01:
> @pavlohamov, is that a regression ?
> If yes, can you point a SHA1/setup that used to work ?

I don't think so. I've just started to work with the wb series. Right now i'm seeking for the ST cube example that implement pairing (seems like they all are with 0 sec level)

joerchan, 2019-10-02:
@pavlohamov Is it possible to get a sniffer capture of the failure? It would make it easier to narrow down the reason.

pavlohamov, 2019-10-02:
@joerchan Do you mean the btmon output? I can do it.
```
CONFIG_BT_DEBUG_MONITOR=y
CONFIG_BT_DEBUG_HCI_CORE=y
CONFIG_BT_DEBUG_HCI_DRIVER=y
```
Is enough?

joerchan, 2019-10-02:
> @joerchan Do you mean the btmon output? I can do it.
> 
> ```
> CONFIG_BT_DEBUG_MONITOR=y
> CONFIG_BT_DEBUG_HCI_CORE=y
> CONFIG_BT_DEBUG_HCI_DRIVER=y
> ```
> 
> Is this log config enough?

It's not what I meant, but you could do that as well.
I would drop the DEBUG_HCI_*, and instead add CONFIG_BT_DEBUG_SMP. 
If you have an extra nrf dev kit you could try the nrf sniffer.
https://www.nordicsemi.com/Software-and-Tools/Development-Tools/nRF-Sniffer-for-Bluetooth-LE/Download#infotabs

pavlohamov, 2019-10-02:
@joerchan 
[smp_fail.log](https://github.com/zephyrproject-rtos/zephyr/files/3680541/smp_fail.log)
Here is the output for only the CONFIG_BT_DEBUG_SMP enabled. I have a nrf52840 dongle. But seems like it's not supported

jhedberg, 2019-10-02:
@pavlohamov the `bt_smp.smp_pairing_complete: status 0xb` in the logs maps to `BT_SMP_ERR_DHKEY_CHECK_FAILED`

pavlohamov, 2019-10-02:
@jhedberg Yeah. i've seen it. For example if remove the passkey display function the same failure will occur (but there is no passkey at all).
[smp_fail_nodisplay.log](https://github.com/zephyrproject-rtos/zephyr/files/3680616/smp_fail_nodisplay.log)

jhedberg, 2019-10-02:
@pavlohamov considering the inputs to the DHKey Check value calculation function (aka `f6()` in the SMP spec) it might e.g. be a difference in understanding what the local and peer address is, since both of these are used as inputs to the function. I'm mainly thinking of the ff:ff:ff:ff.. address you're getting.

pavlohamov, 2019-10-02:
@jhedberg Agree. That was my first clue which i described at the beginning of this issue. Suppose either we have some improper configurations of the HCI stack in the M0 Core, or the actual ST BLE HCI implementation is buggy

joerchan, 2019-10-02:
@pavlohamov Who are you connecting to, and why is their address all FFs?
Maybe turn on CONFIG_BT_DEBUG_HCI_CORE after all, it gives a lot of output but maybe we can find something. Please do pairing without passkey so that the log becomes smaller.

No the nrf52840 dongle is unfortunately not supported.

pavlohamov, 2019-10-02:
@joerchan I've tested with some Chinese android (5.0+) and iPhone 7. Currently i'm using iPhone 7 for the test. I don't think that "other end" device is a troublemaker. 

> 
> 
> and why is their address all FFs?

This is the most important question. If i disable the SMP then i will see the peer address
[no_smp.log](https://github.com/zephyrproject-rtos/zephyr/files/3680778/no_smp.log)
But if the SMP is enabled then no peer address is being seen.
[smp_fail.log](https://github.com/zephyrproject-rtos/zephyr/files/3680781/smp_fail.log)
The same application tested on nRF52832 is working fine

joerchan, 2019-10-02:
@pavlohamov Did you include a different run in no_smp.log?
It still shows SMP debug, and the address is still sometimes all FFs.

The on-air address is input to the security procedures, so if we have the wrong address that would explain the DHKEY check failure.
As the address appears to be 54:7b:29:2c:79:84 (random) then there is our problem.

Where do this FF address come from? Is it given to you by the HCI controller?
For this purpose, can you please update this:
```
	BT_DBG("status 0x%02x handle %u role %u %s", evt->status, handle,
	       evt->role, bt_addr_le_str(&evt->peer_addr));
```
to
```
	BT_DBG("status 0x%02x handle %u role %u %s", evt->status, handle,
	       evt->role, bt_addr_le_str(&evt->peer_addr),
	       bt_addr_le_str(&evt->peer_rpa), bt_addr_le_str(&evt->peer_rpa));
```
Do you have CONFIG_BT_PRIVACY enabled?

joerchan, 2019-10-02:
Ok, your HCI controller is giving you an invalid Enhanced Connection Complete Event. Which is why it works without SMP, because then we disable the Enhanced event, and use the Legacy event instead.

It has put the peer address in the peer_rpa entry instead, this should only be used when the controller has resolved the identity of the peer. Which it is not telling us it has, i.e the peer address is random (0x01), not random-id (0x03). And it has no peer identities loaded (hci_id_add is never called).
Do you have an hci_driver og some sort of stm32 specific code where you could add a workaround?

pavlohamov, 2019-10-02:
@joerchan One more input:
```
+++ b/subsys/bluetooth/host/hci_core.c
-       BT_DBG("status 0x%02x handle %u role %u %s", evt->status, handle,
-              evt->role, bt_addr_le_str(&evt->peer_addr));
+       BT_DBG("status 0x%02x handle %u role %u adr:%s rpa:%s local_rpa:%s", evt->status, handle,
+                      evt->role, bt_addr_le_str(&evt->peer_addr),
+                                  bt_addr_str(&evt->peer_rpa), bt_addr_str(&evt->local_rpa));
+
+       bt_addr_copy(&evt->peer_addr.a, &evt->peer_rpa);
```
And now secure connection could be established (with passkey and without, with privacy and without)
[smp_crutch_ok.log](https://github.com/zephyrproject-rtos/zephyr/files/3681150/smp_crutch_ok.log)

pavlohamov, 2019-10-02:
> 
> Do you have an hci_driver og some sort of stm32 specific code where you could add a workaround?

Hm, suppose the event could be "adjusted" here drivers/bluetooth/hci/ipm_stm32wb.c

joerchan, 2019-10-02:
@pavlohamov
Yeah. That might be best. Unless you have the possebility to actually correct the software in the controller?

Something like this perhaps:
```
if (!bt_addr_cmp(&evt->peer_addr.a, BT_ADDR_NONE)) {
	bt_addr_copy(&evt->peer_addr.a, &evt->peer_rpa);
	evt->peer_addr.addr_type = BT_ADDR_TYPE_RANDOM;
}
```
That might be enough to make it work.
You should check that you can re-connect with privacy enabled though.
I.e connect pair, disconnect, re-connect and re-encrypt.

pavlohamov, 2019-10-02:
@joerchan Thank you for your help. Unfortunate no. As far as i know ST provides BLE stack as encrypted binary with no source code to update it myself. 
I've perfumed couple of tests with PRIVACY enabled and disabled. Seems like everything work fine. connection is being automatically re-established. (Rebooted both device several times).

joerchan, 2019-10-02:
@pavlohamov Alright. It's good to hear that it works.
Do you intend to create a pull request with the workaround?

pavlohamov, 2019-10-03:
@joerchan Yes. Here it is https://github.com/zephyrproject-rtos/zephyr/pull/19568

joerchan, 2019-10-03:
Updated title of issue, and impact for future reference.

erwango, 2019-10-03:
@pavlohamov, @joerchan, @jhedberg thanks for efforts on debugging and solving the issue.
I'll report the issue internally and see if there's something that can be done to solve the issue in a future controller binary release.

commit 82ed1681 (2019-10-05): drivers: ipm_stm32wb: add workaround for conn_complete

Enhanced connection complete message lacks of
the peer address (FF::FF), which leads to pairing/bonding failure.

Fixes: #19509
closed 2019-10-05 by andrewboie