zephyrproject-rtos/zephyr #19523: Can't build fade-led example for any Nordic board

ValentiWorkLearning, 2019-10-01:
**Describe the bug**
After generating ninja with CMake can't buld project for Nordic's devkits, such as:
cmake -GNinja -DBOARD=nrf52840_papyr ..

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -GNinja -DBOARD=nrf52840_papyr ..
3. ninja
4. See error

**Expected behavior**
It's possible to build the fade-led example. Maybe explanation for device-tree syntax need to be added to Zephyr.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Screenshots or console output**
```
PS D:\1-Controllers\1-Diploma\Firmware\nrf52_e73mod\build> cmake -GNinja -DBOARD=nrf52840_papyr ..
Zephyr version: 2.0.99
-- Found PythonInterp: C:/Users/Valenti/AppData/Local/Programs/Python/Python37-32/python.exe (found suitable version "3.7.3", minimum required is "3.4")
-- Selected BOARD nrf52840_papyr
-- Found west: C:/Users/Valenti/AppData/Local/Programs/Python/Python37-32/Scripts/west.exe (found suitable version "0.6.2", minimum required is "0.6.0")
-- Loading C:/Users/Valenti/zephyrproject/zephyr/boards/arm/nrf52840_papyr/nrf52840_papyr.dts as base
-- Overlaying C:/Users/Valenti/zephyrproject/zephyr/dts/common/common.dts
Devicetree configuration written to D:/1-Controllers/1-Diploma/Firmware/nrf52_e73mod/build/zephyr/include/generated/generated_dts_board.conf
Parsing Kconfig tree in C:/Users/Valenti/zephyrproject/zephyr/Kconfig
Loaded configuration 'C:/Users/Valenti/zephyrproject/zephyr/boards/arm/nrf52840_papyr/nrf52840_papyr_defconfig'
Merged configuration 'D:/1-Controllers/1-Diploma/Firmware/nrf52_e73mod/prj.conf'
Configuration saved to 'D:/1-Controllers/1-Diploma/Firmware/nrf52_e73mod/build/zephyr/.config'
-- Cache files will be written to: C:\Users\Valenti\AppData\Local/.cache/zephyr
-- The C compiler identification is GNU 8.3.1
-- The CXX compiler identification is GNU 8.3.1
-- The ASM compiler identification is GNU
-- Found assembler: C:/gcc_none_eabi/bin/arm-none-eabi-gcc.exe
-- Configuring done
-- Generating done
-- Build files have been written to: D:/1-Controllers/1-Diploma/Firmware/nrf52_e73mod/build
PS D:\1-Controllers\1-Diploma\Firmware\nrf52_e73mod\build> ninja
[8/125] Building C object CMakeFiles/app.dir/src/main.c.obj
FAILED: CMakeFiles/app.dir/src/main.c.obj
C:\gcc_none_eabi\bin\arm-none-eabi-gcc.exe -DBUILD_VERSION=zephyr-v2.0.0-801-g5639ea07f8dd -DKERNEL -DNRF52840_XXAA -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR__=1 -IC:/Users/Valenti/zephyrproject/zephyr/kernel/include -IC:/Users/Valenti/zephyrproject/zephyr/arch/arm/include -IC:/Users/Valenti/zephyrproject/zephyr/include -IC:/Users/Valenti/zephyrproject/zephyr/include/drivers -Izephyr/include/generated -IC:/Users/Valenti/zephyrproject/zephyr/soc/arm/nordic_nrf/nrf52 -IC:/Users/Valenti/zephyrproject/zephyr/ext/hal/cmsis/Include -IC:/Users/Valenti/zephyrproject/modules/hal/nordic/nrfx -IC:/Users/Valenti/zephyrproject/modules/hal/nordic/nrfx/drivers/include -IC:/Users/Valenti/zephyrproject/modules/hal/nordic/nrfx/mdk -IC:/Users/Valenti/zephyrproject/modules/hal/nordic/. -isystem C:/Users/Valenti/zephyrproject/zephyr/lib/libc/minimal/include -isystem c:/gcc_none_eabi/bin/../lib/gcc/arm-none-eabi/8.3.1/include -isystem c:/gcc_none_eabi/bin/../lib/gcc/arm-none-eabi/8.3.1/include-fixed -Os -imacrosD:/1-Controllers/1-Diploma/Firmware/nrf52_e73mod/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -mthumb -mcpu=cortex-m4 -imacrosC:/Users/Valenti/zephyrproject/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=D:/1-Controllers/1-Diploma/Firmware/nrf52_e73mod=CMAKE_SOURCE_DIR -fmacro-prefix-map=C:/Users/Valenti/zephyrproject/zephyr=ZEPHYR_BASE -ffunction-sections -fdata-sections -mabi=aapcs -march=armv7e-m -std=c99 -nostdinc -MD -MT CMakeFiles/app.dir/src/main.c.obj -MF CMakeFiles\app.dir\src\main.c.obj.d -o CMakeFiles/app.dir/src/main.c.obj   -c ../src/main.c
../src/main.c:23:2: error: #error "Choose supported PWM driver"
 #error "Choose supported PWM driver"
  ^~~~~
../src/main.c: In function 'main':
../src/main.c:43:31: error: 'PWM_DRIVER' undeclared (first use in this function); did you mean 'PWM_0_SIZE'?
  pwm_dev = device_get_binding(PWM_DRIVER);
                               ^~~~~~~~~~
                               PWM_0_SIZE
../src/main.c:43:31: note: each undeclared identifier is reported only once for each function it appears in
../src/main.c:50:33: error: 'PWM_CHANNEL' undeclared (first use in this function); did you mean 'PWM0_CH_NUM'?
   if (pwm_pin_set_usec(pwm_dev, PWM_CHANNEL,
                                 ^~~~~~~~~~~
                                 PWM0_CH_NUM
[13/125] Building C object zephyr/arch/arch/arm/core/CMakeFiles/arch__arm__core.dir/nmi.c.obj
ninja: build stopped: subcommand failed.
PS D:\1-Controllers\1-Diploma\Firmware\nrf52_e73mod\build>
```
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain: gnu-arm-none-eabi
 - Commit - latest avaliable version

**Additional context**
I tried to bulid the fade-led example for custom Nordic board and see this error about non-aliased pwe0 led controller. Later I found, that this example couldn't be built for any of Nordic's boards.

carlescufi, 2019-10-01:
Changed from bug to  question since this is likely just Device Tree. I leave it to @anangl to confirm.

ValentiWorkLearning, 2019-10-01:
@carlescufi Thanks! Awaiting for answer.

ValentiWorkLearning, 2019-10-01:
FYI: https://github.com/zephyrproject-rtos/zephyr/issues/19467

anangl, 2019-10-02:
@carlescufi 
Yes, this is a question of devicetree - Nordic boards lack definitions of `pwmleds` and the `pwm-led0` alias that are required by the fade_led sample. But it is actually a bug because fade_led's `README.rst` still states that the sample should work on nrf52840_pca10056 "out of the box".
I think I'll manage to come soon (meaning: tomorrow) with a PR that will correct this for Nordic DK boards. And that PR could be a base for all users wanting to update definitions for any other board with a Nordic SoC.

mron, 2019-10-02:
Thanks for looking into this. Newbie question:
How will I use the PR ( Pull Request? )? Pointing me to the right GitHub doc would be great.

ValentiWorkLearning, 2019-10-03:
@anangl Where I can find detailed information about device-tree structure and how to use aliases/add new peripheral? Is this available in Zephyr docs?

anangl, 2019-10-03:
@ValentiWorkLearning Look here: https://docs.zephyrproject.org/latest/guides/dts/index.html

mron, 2019-10-03:
I downloaded anangel's changes ( using svn ) and tried them out with board=nrf52840_pca10056. It compiled without errors.

Thanks!

closed 2020-10-04 by github-actions[bot]