zephyrproject-rtos/zephyr #19527: LwM2M socket management to allow device power management

tautologyclub, 2019-10-01:
I don't think there is an issue for implementing lwm2m queue mode, but it's possible such an issue would either overlap, or have dependencies upon, this issue. @mike-scott knows more than I do.

Aaaaaaaanyway. Currently, the lwm2m engine spawns a `k_thread` which basically performs a very tight `poll`/`recv`/`k_sleep` loop, for infinity. This means:

1) the system can't ever gate its HF clock and low power idle
2) we can never power off the radio

Clearly this is also somewhat related to https://github.com/zephyrproject-rtos/zephyr/issues/19526 which would give the application freedom to implement power management as part of its own connection management implementation.

dhandzic, 2021-02-24:
We are using LWM2M on a nRF9160 and experiencing much higher current consumption than expected. We think this is related to the application processor never entering any power saving mode.
What is the status for this issue?

rlubos, 2021-02-24:
No progress yet.

dhandzic, 2021-03-01:
Could you direct me to where this problematic code is located? Maybe I can have look at it and contribute some improvement.

rlubos, 2021-03-01:
The issue refers to the main loop of the lwm2m_engine:
https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/net/lib/lwm2m/lwm2m_engine.c#L4614

I think we should try to aim to have an API similar to the MQTT library in Zephyr, where we have the `mqtt_input()` function to process any incoming data, `mqtt_live()` to send keep-alive messages when needed and `mqtt_keepalive_time_left()` to determine how long can the MQTT lib reamain idle (in case there's no incoming data).

With LwM2M it's a bit more complex though, as the engine is not only responsible for sending the keep-alive messages (Register Update), but also for handling observations and any other "services" that are registered (AFAIR the Device object registers such "service"). I don't have a clear vision on how to handle that yet.

dhandzic, 2021-03-25:
I think one solution could be to observe the radio and evaluate the defined objects. If eDRX is used no incoming messages will arrive during the eDRX cycles "off" period and the LWM2M main loop could go to idle as well. If there are no objects with Observe option = 0. The LWM2M client will only need to respond to incoming messages. If there are objects with Observe the notification will not be sent more often than once every **pmin** seconds and if pmax is a non-zero value the notification shall be sent every **pmax** seconds even if the value did not change .

qlocx, 2021-03-26:
I'm very interested in a solution to this issue since my project is observing the same problems.. Do anyone have any further input?

endian-benjamin, 2021-06-02:
@rlubos Isn't queue mode + dtls connection id the answer?

dhandzic, 2021-06-02:
> 
> 
> @rlubos Isn't queue mode + dtls connection id the answer?

I dont think queue mode is the answer, since it means the device does not need to listen all the time. This is not a feasible solution as soon as the device is used to actuate anything and reasonably fast response is required. The sleeping time of the LWM2M engine needs rather to be synchronized with the modem. The latest modem firmware 1.30 for i e nRF9160 offers AT commands to let the modem notify the application about its state. If the modem enters eDRX or PSM low power mode the LWM2M engine can sleep as well.

kartben, 2023-08-20:
Is this still an issue now that tickless mode has been added?
@rlubos @fzdobylak

fzdobylak, 2023-08-21:
@kartben I don't think I have any expertise in that topic.

kartben, 2023-08-21:
> @kartben I don't think I have any expertise in that topic.

woops sorry, I meant to tag @SeppoTakalo

rlubos, 2023-08-21:
Hmm, I think that tickless mode, combined with `lwm2m_engine_pause()/resume()` APIs should allow for efficient power management of LwM2M applications. Therefore I think we can close the issue.

closed 2023-08-21 by rlubos