zephyrproject-rtos/zephyr #19530: Enhance sanitycheck/CI for separating build phase from testing

galak, 2019-10-01:
We should look at enhancements to allow separation of building tests v running them.  With the changes in PR #18603 we introduce a --test-only option.  This works reasonably well on a local system.

We should provide an option in sanitycheck to create the bare minimum artifacts to allow running tests at some later stage.  The current minimum artifacts would be the zephyr binaries, a script that has the commands to do simulation runs, and the test/sample.yaml to provide any handler information.

This would allow for us to separate the build phase from the run phase for CI of PR jobs.  This would also allow us to publish the builds from nightly and let users take the build artifacts and report run results to testrail.

marc-hb, 2019-10-01:
> in PR #18603 we introduce a --test-only option.

= "Only run device tests with current artifacts, do not build the code"

Is there already a way to exclude "build only" configurations to build *and* run the others? (That's how I initially misunderstood `--test-only`)

hakehuang, 2019-10-02:
@mar-hb would this option meet your requirement?
`
-b, --build-only      Only build the code, do not execute any of it in QEMU
`

galak, 2019-10-02:
> > in PR #18603 we introduce a --test-only option.
> 
> = "Only run device tests with current artifacts, do not build the code"
> 
> Is there already a way to exclude "build only" configurations to build _and_ run the others? (That's how I initially misunderstood `--test-only`)

Hmm, not sure how @nashif meant that.  I took it to mean run the tests only phase.

marc-hb, 2019-10-02:
Yes, I understood eventually. I'm just wondering whether the other, test filtering feature is already there somehow or should I file a new issue for it.

galak, 2019-10-02:
> Yes, I understood eventually. I'm just wondering whether the other, test filtering feature is already there somehow or should I file a new issue for it.

I feel like @nashif made a change along those lines already.

nashif, 2020-11-28:
--build-only and --test-only now available and supported.

closed 2020-11-28 by nashif