zephyrproject-rtos/zephyr #19537: debug:object_tracing: The trace list is not complete once we initialize the object on the trace list

Teng-Shih-Wei, 2019-10-02:
Problem statement:
The scenario of demo "mutex_api" will reused the mutex/kmutex. It will initialize them many times, and I found the objects are lost.

Root cause:
When we initialize a kernel object(e.g: mutex), it will do "SYS_TRACEING_OBJECT_INIT" which link
the object to the list head. If the object is in the middle of the trace list and we initialize it again,
the list tail will be lost and the trace list become an incomplete circular lined list.

Solution proposal:
I'd like to add a flag "__linked" to identify whether this object is linked in the trace list. 
Add a condition in macro "SYS_TRACEING_OBJECT_INIT" to check the flag. If it is not in the 
list, add it to the list head. Otherwise, do nothing.

mped-oticon, 2019-10-02:
@Teng-Shih-Wei This is for the SystemView Segger RTT backend of tracing, right?
It would be great if your proposed flag could remain within the RTT backend.

I do not know much about RTT, but I gather there are two circular queues, up and down. And you have problem with the Up buffer.

Teng-Shih-Wei, 2019-10-04:
@mped-oticon I read the SEGGER SystemView user guild 10.1.3. For getting task information, 
developer has to implement an callback function. This callback function need to traverse the task list 
and collect the information of each task. 

I think the way to collect IPC information is the same, so adding a flag in IPC structure should be fine.

Teng-Shih-Wei, 2019-10-07:
@mped-oticon I am not using SystemView actually. I found this issue when I ran the test case 
 "mutex_api". This issue is for the raw kernel information, so any backend of debugging tool may have 
 problem due to this issue.

BTW, I visit SystemView website, and found that the the kernel information is encapsulated it in RTT 
control block, put in up buffer, and wait for being transferred to the host. This issue will make the data in up buffer incorrect. Besides, the RTT backend may not stop due to the circular list.

PR #19675 (2019-11-14): object tracing: Fix the issue that objects lost from trace list
Add a flag for identifing whether this object is on the trace
list. Ensure that link any object to the trace list only one time.
It will avoid the issue about lost object caused by adding a
object to trace list twice.

Fixes #19537
commit 5ebceeb8 (2019-11-14): object tracing: Fix the issue that objects lost from trace list

Add a flag for identifing whether this object is on the trace
list. Ensure that link any object to the trace list only one time.
It will avoid the issue about lost object caused by adding a
object to trace list twice.

Fixes #19537
closed 2019-11-14 by nashif