zephyrproject-rtos/zephyr #19540: Allow running and testing network samples in automatic way

jukkar, 2019-10-02:
**Is your enhancement proposal related to a problem? Please describe.**
The network samples can bit rot easily as they are only compile tested by sanitycheck script.
We need to run the network samples in some semi-automatic way so that we can be sure that there is no regressions in network stack.

**Describe the solution you'd like**
Use Docker to to run scripts and programs from net-tools, then have qemu / native_posix to connect to that Docker instance. Have also IP addresses setup properly so that connectivity between these two works.

dleach02, 2019-11-05:
@jukkar, I note that all the subtasks are checked off but it is unclear to me if there are actual PRs or items merged into main?

pfl, 2019-11-05:
>  note that all the subtasks are checked off but it is unclear to me if there are actual PRs or items merged into main?

We have https://github.com/zephyrproject-rtos/zephyr/pull/19677 in the pipeline implementing the ticked off items. The support in net-tools was handled by https://github.com/zephyrproject-rtos/net-tools/pull/37 which was already merged.

commit 65ffc714 (2020-02-03): scripts: net: Add script for running sample tests

Add a script that sets up Docker networking and starts the net-tools
Docker container. If successful, run Zephyr with native_posix and
execute the appropriate net-tools container executable. The proper
net-tools executable and arguments is selected depending on the
basename of the sample directory. This script needs to be updated
if the net-tools Docker image is updated in some incompatible way.

The net-tools directory is assumed to exist at the same level as
the Zephyr base directory, but its location can be set using the
'-N' command line a [... cut]
closed 2020-02-03 by nashif