zephyrproject-rtos/zephyr #19543: net: tcp: echo server stops if CONFIG_POSIX_MAX_FDS is not set

rmstoi, 2019-10-02:
If CONFIG_POSIX_MAX_FDS is not set or commented out in prj.conf echo server stops when receiving basic TCP SYN:

```
[00:00:11.760,000] <inf> net_echo_server_sample: Waiting for TCP connection on port 4242 (IPv4)...
[00:00:11.760,000] <err> net_echo_server_sample: Error in accept (IPv4): -23 - stopping server
[00:00:11.760,000] <inf> net_echo_server_sample: Stopping...
```

[tcpsyn.pcap.gz](https://github.com/zephyrproject-rtos/zephyr/files/3680046/tcpsyn.pcap.gz)

jukkar, 2019-10-02:
The culprit is probably the fact that the value is too low, by default it is 4 which is too low if multiple listeners are enabled in echo-server.

jukkar, 2019-10-18:
There is not much we can do in this case, the application does not have enough resources to continue and it will exit.